bambulab/BambuStudio #5132: Tree supports dont generate top interface layers

didgeridoooo, 2024-11-03:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Insert model
Enable Support
Select tree(auto)
Set top interface spacing to 0
Slice model

### Actual results

Top interface layers are just the top of the tree supports, with no bridging between supports before first model layer. This results in terrible model layers above the tree supports
[image]
[image]

### Expected results

Top interface layer should be solid, like it was in version 1.8.4.
[image]

### Project file & Debug log uploads

[project.zip](https://github.com/user-attachments/files/17609447/Handle.zip)
[log.zip](https://github.com/user-attachments/files/17609448/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

didgeridoooo, 2024-11-07:
I posted here and to customer support because I wasn't sure if it was a bug or not. I am including the response from customer support for anyone looking for a workaround right now. Turning off variable layer height did in fact generate support interface correctly.

_Through our testing and verification, we have indeed noticed that Bambu Studio versions after 1.8.4 do not generate a complete support interface. The cause of this issue is related to changes in the logic for variable layer height and tree supports in the new version of Bambu Studio. We observed that your model contains variable layer height, which may affect the generation logic for tree supports in this situation.
 
To address your current issue, we recommend disabling the variable layer height or temporarily using an older version of Bambu Studio to enable the use of both variable layer height and multi-layer support surfaces with tree supports.
 
We appreciate your understanding and cooperation, and we apologize for any inconvenience you may have experienced.
 
Best regards,
Bambu Lab Customer Support_

praetp, 2024-11-08:
I confirm it works 1.8.4 but I did not see it working again in recent versions when disabling variable layer height...

tangketan, 2024-11-14:
Yes, it is a problem. We'll try to bring back the complete interface in the next version.

gileseperon, 2025-05-13:
this still seems like a problem for me too.

JustinBrown17, 2025-11-08:
Still an issue. I found changing the branch distance to a small number (like 1mm) seems to give the full support interface, though this is a work around.. Hopefully this helps someone till this gets resolved.