bambulab/BambuStudio #5133: Internal solid filling.

Tony3DPrinting, 2024-11-03:
Bambu Studio v. 
Why does the internal solid fill texture not change but is always Gyroid?
Even changing texture this always remains Gyroid.
Thanks.

jm355, 2024-11-11:
if you want a solid object, you have to set the sparse infill pattern to rectilinear and the sparse infill density to 100%. There are other ways to do it but that's the easiest I know of

Tony3DPrinting, 2024-11-12:
E chi ha parlato di questo?
Io non ho chiesto questo.
Ho già parlato del problema riscontrato.
Come ho già detto ho notato che con alcune forme e dimensioni il pattern del riempimento solido dopo il primo layer (non riempimento sparso) anche se si va a cambiare resta sempre concentrico.
Spero che almeno questa volta di essere stato chiaro.