bambulab/BambuStudio #5135: 1.10 Beta: Crashing when using cross-subnet LAN-only mode

SlothCroissant, 2024-11-04:
### Bambu Studio Version

1.10.0.81

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS Sonoma 14.2.1

### Additional system information

MacBook Pro 
CPU: M1 Pro
RAM: 16GB

### Printer

Bambu P1P

### How to reproduce

1. Open Bambu Studio Beta
2. Devices Tab
3. Add Device
4. Enter IP Address & Access Code

### Actual results

Application crashes

### Expected results

Application connects to LAN mode printer

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17612766/log.zip)

No project file involved.

### Checklist of files to include

- [X] Log file
- [ ] Project file

SlothCroissant, 2024-11-04:
Apple crash log:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [89448]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.10.00.81 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-11-03 18:31:26.9246 -0600
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        F358C3AD-FE94-C441-544B-0C78869A8BAE

Sleep/Wake UUID:       18BA0C57-B5F7-4F0E-892E-FB049FC2CEED

Time Awake Since Boot: 62000 seconds
Time Since Wake:       33783 seconds

System Integrity Protection: enabled

Crashed Thread:        15

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   BambuStudio [89448]

Application Specific Information:
abort() called

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   CoreFoundation                	       0x189a53f38 _CFStringGetCStringPtrInternal + 276
1   CoreFoundation                	       0x189a5d258 CFStringCompareWithOptionsAndLocale + 352
2   HIToolbox                     	       0x1940703c0 RunCurrentEventLoopInMode + 156
3   HIToolbox                     	       0x1940700d8 ReceiveNextEventCommon + 220
4   HIToolbox                     	       0x19406ffdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
5   AppKit                        	       0x18d29e8a4 _DPSNextEvent + 660
6   AppKit                        	       0x18da78980 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
7   AppKit                        	       0x18d4d209c -[NSApplication _doModalLoop:peek:] + 216
8   AppKit                        	       0x18d6c7504 __33-[NSApplication runModalSession:]_block_invoke_2 + 88
9   AppKit                        	       0x18d6c7490 __33-[NSApplication runModalSession:]_block_invoke + 112
10  AppKit                        	       0x18d4d081c _NSTryRunModal + 100
11  AppKit                        	       0x18d6c7360 -[NSApplication runModalSession:] + 148
12  BambuStudio                   	       0x10391cb44 wxGUIEventLoop::DoDispatchTimeout(unsigned long) + 64
13  BambuStudio                   	       0x1039f5698 wxCFEventLoop::DispatchTimeout(unsigned long) + 44
14  BambuStudio                   	       0x1039f57b8 wxCFEventLoop::OSXDoRun() + 172
15  BambuStudio                   	       0x10391d088 wxModalEventLoop::OSXDoRun() + 88
16  BambuStudio                   	       0x1039f5874 wxCFEventLoop::DoRun() + 40
17  BambuStudio                   	       0x103974f30 wxEventLoopBase::Run() + 132
18  BambuStudio                   	       0x1038abd8c wxDialog::ShowModal() + 100
19  BambuStudio                   	       0x101aa6088 Slic3r::GUI::SelectMachinePopup::OnLeftUp(wxMouseEvent&) + 1184
20  BambuStudio                   	       0x1039ec980 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
21  BambuStudio                   	       0x1039ec660 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
22  BambuStudio                   	       0x1039ec484 wxEvtHandler::ProcessEvent(wxEvent&) + 56
23  BambuStudio                   	       0x10382ead8 wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) + 48
24  BambuStudio                   	       0x1039ec6ec wxEvtHandler::ProcessEventLocally(wxEvent&) + 228
25  BambuStudio                   	       0x1039ec484 wxEvtHandler::ProcessEvent(wxEvent&) + 56
26  BambuStudio                   	       0x1039ecb2c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
27  BambuStudio                   	       0x10393b26c wxWidgetCocoaImpl::DoHandleMouseEvent(NSEvent*) + 92
28  BambuStudio                   	       0x103924b38 -[NSWindow(wxNSWindowSupport) WX_filterSendEvent:] + 168
29  BambuStudio                   	       0x10392502c -[wxNSPanel sendEvent:] + 36
30  AppKit                        	       0x18da779f4 -[NSApplication(NSEventRouting) sendEvent:] + 1604
[... 535 lines cut ...]
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

Model: MacBookPro18,3, BootROM 10151.61.4, proc 8:6:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  1 2023 19:46:07 version 20.10.1073.3.8.7.165 FWID 01-b66346c5
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: YubiKey FIDO+CCID
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

walterwongbbl, 2024-11-12:
@SlothCroissant 
We have fixed this issue on the macOS system. There will be a new version later on

SlothCroissant, 2024-12-06:
> @SlothCroissant We have fixed this issue on the macOS system. There will be a new version later on

Thanks @walterwongbbl - Checked on the recent release and don't see indication of this one. Can you confirm which release this fix will be in?

Thanks!

bandesz, 2024-12-22:
1.10.1.50 doesn't crash for me, but the different Wifi networks are causing an issue: https://github.com/bambulab/BambuStudio/issues/4202