bambulab/BambuStudio #5136: Arachne wall generator creates 2 layers at 1M+ mm height for the tips of cones

Narvath, 2024-11-04:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Not a perfomance issue

### Printer

A1 mini, 0.4mm nozzle

### How to reproduce

Open the provided Caliper_Cones_Arachne.3mf and slice

### Actual results

The last two layers are at 1M+ mm of height. See attachment : 
[image]

Also worth noting is that these two layers apear at the right height in the preview tab.

### Expected results

It should stop at the layer 270 as in this screenshot, result from the Caliper_Cones_Classic.3mf :
[image]

### Project file & Debug log uploads

Here are the log that should contain the "A G-code path goes beyond the max print height" related to the issue.
[log.zip](https://github.com/user-attachments/files/17612765/log.zip)

Here are the two 3mf files, one with the classic generator which gives the intended result and the arachne one which creates the two out of bound layers.
[3mf_Files.zip](https://github.com/user-attachments/files/17612767/3mf_Files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Narvath, 2024-11-05:
Still says that 1.9.7.52 is the latest available version, I'm guessing that is due because that is the version from the official website?

QingZhangBambu, 2024-11-06:
@Narvath the beta version has been released, or you can wait for 1.10 official in one or two weeks
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81