bambulab/BambuStudio #5137: Text shape no longer supports any but the default font

cdubs-cyber, 2024-11-04:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.1

### Printer

Bambu Labs X1 Carbon

### How to reproduce

1. Open a model
2. Select the "Text Shape" tool
3. Change the font to any font besides the default
4. Type text to create the shape
5. Font will stay as default, and will not change to the selected font

### Actual results

The text will not change to the selected font, but will stay the default font.

[image]

### Expected results

The font should change to the selected font.

### Project file & Debug log uploads

[MapleSMALL_bug.3mf.zip](https://github.com/user-attachments/files/17613399/MapleSMALL_bug.3mf.zip)
debug_Sun_Nov_03_19_47_23_83848.log.0

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-11-04:
Hello, we are currently working on resolving this issue.

adamurban, 2025-01-04:
@Haidiye00 any update on this one?

Here is the error I see in my console when trying to load a font:

`Font_FontMgr, warning: unable to find font 'Apple Chancery' [bold]; 'Arial' [aspects: regular,bold,italic,bold-italic] [paths: /System/Library/Fonts/Supplemental/Arial.ttf;/System/Library/Fonts/Supplemental/Arial Bold.ttf;/System/Library/Fonts/Supplemental/Arial Italic.ttf;/System/Library/Fonts/Supplemental/Arial Bold Italic.ttf] is used instead`

Of course, the specified font is indeed present on the filesystem, so it seems like a bug in the open cascade font manager library you are using?

`% ls /System/Library/Fonts/Supplemental/                     
Academy Engraved LET Fonts.ttf			Muna.ttc					NotoSansTifinagh-Regular.otf
Al Nile.ttc					Myanmar MN.ttc					NotoSansTirhuta-Regular.ttf
Al Tarikh.ttc					Myanmar Sangam MN.ttc				NotoSansUgaritic-Regular.ttf
AlBayan.ttc					NISC18030.ttf					NotoSansVai-Regular.ttf
AmericanTypewriter.ttc				Nadeem.ttc					NotoSansWancho-Regular.ttf
Andale Mono.ttf					NewPeninimMT.ttc				NotoSansWarangCiti-Regular.ttf
Apple Chancery.ttf				NotoSansAdlam-Regular.ttf			NotoSansYi-Regular.ttf
AppleGothic.ttf					NotoSansAvestan-Regular.ttf			NotoSerifAhom-Regular.ttf
AppleMyungjo.ttf				NotoSansBamum-Regular.ttf			NotoSerifBalinese-Regular.ttf
Arial Black.ttf					NotoSansBassaVah-Regular.ttf			NotoSerifNyiakengPuachueHmong-Regular.ttf
Arial Bold Italic.ttf				NotoSansBatak-Regular.ttf			NotoSerifYezidi-Regular.otf
Arial Bold.ttf					NotoSansBhaiksuki-Regular.ttf			Oriya MN.ttc
Arial Italic.ttf				NotoSansBrahmi-Regular.ttf			Oriya Sangam MN.ttc
Arial Narrow Bold Italic.ttf			NotoSansBuginese-Regular.ttf			PTMono.ttc
Arial Narrow Bold.ttf				NotoSansBuhid-Regular.ttf			PTSans.ttc
Arial Narrow Italic.ttf				NotoSansCanadianAboriginal-Regular.otf		PTSerif.ttc
Arial Narrow.ttf				NotoSansCarian-Regular.ttf			PTSerifCaption.ttc
Arial Rounded Bold.ttf				NotoSansCaucasianAlbanian-Regular.ttf		Papyrus.ttc
Arial Unicode.ttf				NotoSansChakma-Regular.ttf			PartyLET-plain.ttf
Arial.ttf					NotoSansCham-Regular.ttf			Phosphate.ttc
Athelas.ttc					NotoSansCoptic-Regular.ttf			PlantagenetCherokee.ttf
Ayuthaya.ttf					NotoSansCuneiform-Regular.ttf			Raanana.ttc
Baghdad.ttc					NotoSansCypriot-Regular.ttf			Rockwell.ttc
Bangla MN.ttc					NotoSansDuployan-Regular.ttf			STIXGeneral.otf
Bangla Sangam MN.ttc				NotoSansEgyptianHieroglyphs-Regular.ttf		STIXGeneralBol.otf
Baskerville.ttc					NotoSansElbasan-Regular.ttf			STIXGeneralBolIta.otf
Beirut.ttc					NotoSansGlagolitic-Regular.ttf			STIXGeneralItalic.otf
BigCaslon.ttf					NotoSansGothic-Regular.ttf			STIXIntDBol.otf
Bodoni 72 OS.ttc				NotoSansGunjalaGondi-Regular.otf		STIXIntDReg.otf
Bodoni 72 Smallcaps Book.ttf			NotoSansHanifiRohingya-Regular.ttf		STIXIntSmBol.otf
Bodoni 72.ttc					NotoSansHanunoo-Regular.ttf			STIXIntSmReg.otf
Bodoni Ornaments.ttf				NotoSansHatran-Regular.ttf			STIXIntUpBol.otf
Bradley Hand Bold.ttf				NotoSansImperialAramaic-Regular.ttf		STIXIntUpDBol.otf
Brush Script.ttf				NotoSansInscriptionalPahlavi-Regular.ttf	STIXIntUpDReg.otf
Chalkboard.ttc					NotoSansInscriptionalParthian-Regular.ttf	STIXIntUpReg.otf
ChalkboardSE.ttc				NotoSansJavanese-Regular.otf			STIXIntUpSmBol.otf
Chalkduster.ttf					NotoSansKaithi-Regular.ttf			STIXIntUpSmReg.otf
Charter.ttc					NotoSansKayahLi-Regular.ttf			STIXNonUni.otf
Cochin.ttc					NotoSansKharoshthi-Regular.ttf			STIXNonUniBol.otf
Comic Sans MS Bold.ttf				NotoSansKhojki-Regular.ttf			STIXNonUniBolIta.otf
Comic Sans MS.ttf				NotoSansKhudawadi-Regular.ttf			STIXNonUniIta.otf
Copperplate.ttc					NotoSansLepcha-Regular.ttf			STIXSizFiveSymReg.otf
Corsiva.ttc					NotoSansLimbu-Regular.ttf			STIXSizFourSymBol.otf
Courier New Bold Italic.ttf			NotoSansLinearA-Regular.ttf			STIXSizFourSymReg.otf
Courier New Bold.ttf				NotoSansLinearB-Regular.ttf			STIXSizOneSymBol.otf
Courier New Italic.ttf				NotoSansLisu-Regular.ttf			STIXSizOneSymReg.otf
Courier New.ttf					NotoSansLycian-Regular.ttf			STIXSizThreeSymBol.otf
DIN Alternate Bold.ttf				NotoSansLydian-Regular.ttf			STIXSizThreeSymReg.otf
DIN Condensed Bold.ttf				NotoSansMahajani-Regular.ttf			STIXSizTwoSymBol.otf
Damascus.ttc					NotoSansMandaic-Regular.ttf			STIXSizTwoSymReg.otf
DecoTypeNaskh.ttc				NotoSansManichaean-Regular.ttf			STIXTwoMath.otf
Devanagari Sangam MN.ttc			NotoSansMarchen-Regular.ttf			STIXTwoText-Italic.ttf
DevanagariMT.ttc				NotoSansMasaramGondi-Regular.otf		STIXTwoText.ttf
Didot.ttc					NotoSansMeeteiMayek-Regular.ttf			STIXVar.otf
Diwan Kufi.ttc					NotoSansMendeKikakui-Regular.ttf		STIXVarBol.otf
Diwan Thuluth.ttf				NotoSansMeroitic-Regular.ttf			Sana.ttc
EuphemiaCAS.ttc					NotoSansMiao-Regular.ttf			Sathu.ttf
Farah.ttc					NotoSansModi-Regular.ttf			Savoye LET.ttc
Farisi.ttf					NotoSansMongolian-Regular.ttf			Seravek.ttc
Futura.ttc					NotoSansMro-Regular.ttf				Shree714.ttc
Galvji.ttc					NotoSansMultani-Regular.ttf			SignPainter.ttc
[... 11 lines cut ...]
Hoefler Text Ornaments.ttf			NotoSansOldTurkic-Regular.ttf			Tamil Sangam MN.ttc
Hoefler Text.ttc				NotoSansOsage-Regular.ttf			Telugu MN.ttc
ITFDevanagari.ttc				NotoSansOsmanya-Regular.ttf			Telugu Sangam MN.ttc
Impact.ttf					NotoSansPahawhHmong-Regular.ttf			Thonburi.ttc
InaiMathi-MN.ttc				NotoSansPalmyrene-Regular.ttf			Times New Roman Bold Italic.ttf
Iowan Old Style.ttc				NotoSansPauCinHau-Regular.ttf			Times New Roman Bold.ttf
Kailasa.ttc					NotoSansPhagsPa-Regular.ttf			Times New Roman Italic.ttf
[... cut ...]
Lao Sangam MN.ttf				NotoSansSylotiNagri-Regular.ttf			Verdana.ttf
Luminari.ttf					NotoSansSyriac-Regular.ttf			Waseem.ttc
Malayalam MN.ttc				NotoSansTagalog-Regular.ttf			Webdings.ttf
Malayalam Sangam MN.ttc				NotoSansTagbanwa-Regular.ttf			Wingdings 2.ttf
Marion.ttc					NotoSansTaiLe-Regular.ttf			Wingdings 3.ttf
Microsoft Sans Serif.ttf			NotoSansTaiTham-Regular.ttf			Wingdings.ttf
Mishafi Gold.ttf				NotoSansTaiViet-Regular.ttf			Zapfino.ttf
Mishafi.ttf					NotoSansTakri-Regular.ttf
Mshtakan.ttc					NotoSansThaana-Regular.ttf`