bambulab/BambuStudio #5140: Surface Detection Issue with Roof Components in Slicer – Misaligned Wall Layers and Wave Patterns in Print

butimo, 2024-11-04:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1

### Additional system information

MacBook Pro, 2021, Apple M1 Pro

### Printer

X1C and A1

### How to reproduce

There appears to be an issue with the slicer’s ability to correctly identify flat surfaces on certain roof components, resulting in misaligned wall layers and print quality issues. The problem occurs with two specific roof parts:

1. **Round Roof Component with Flat Dome**: The slicer seems unable to correctly interpret the flat surface of the dome as an outer surface. Instead of forming a consistent, flat layer, the slicer shifts walls upwards, causing the internal walls to become visible through the outer layers. This issue results in wave patterns in the printed filament, especially at the misaligned wall segments.
2. **Kidney-Shaped Roof with Flat Slopes**: Similarly, on this component, the slicer misinterprets the flat roof slopes, failing to recognize them as surfaces. This misinterpretation leads to unpredictable changes in wall alignment, and the filament forms visible waves on these parts during printing. In certain areas, the slicer inexplicably changes its interpretation of the surface mid-roof, creating inconsistent layer patterns.

These issues not only disrupt the visual appearance of the prints but also impact their structural quality by creating weak points. Both components show visible defects where the filament lays unevenly due to the incorrect wall and surface interpretations.

**Steps to Reproduce:**

1. load the attached 3mf file (reduced).
2. Select X1 or A1 with 0.4 nozzle printer
3. Select 0.20mm Standard @BBL global preset
4. Slice model

### Actual results

The slicer misinterprets these top surfaces. You see yellow/orange wall lines instead of red surface lines. Suddenly in some parts is switches from surface to walls. You see the yellow inner walls through the top roof. During printing in those areas it also creates waves (not allways but allways in those areas without top surface).

[image]
[image]
[image]
[image]

### Expected results

The slicer should recognize the flat dome and flat slopes as top surfaces (red) and create consistent roofs. This should prevent visible internal walls and minimize filament wave formations in the final print. 

### Project file & Debug log uploads

[bbl-bug-report.3mf.zip](https://github.com/user-attachments/files/17617483/bbl-bug-report.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

butimo, 2024-11-04:
No idea if there is any thing in the log, which is helpful.

[debug_Mon_Nov_04_13_59_29_17943.log.0.zip](https://github.com/user-attachments/files/17618406/debug_Mon_Nov_04_13_59_29_17943.log.0.zip)

QingZhangBambu, 2024-11-15:
@butimo you can adjust the threshold to control it.
[image]

butimo, 2024-11-15:
@QingZhangBambu thanks, will give it a try. 
It's worth mentioning that this is a "devmode" feature. Took me quite a while to find it ;-)

QingZhangBambu, 2024-11-15:
@butimo yep, I forget that.

butimo, 2024-11-18:
@QingZhangBambu did some testing. Now it came out as expected. Thanks for pointing me to this settings.