bambulab/BambuStudio #5142: unable to login in bambu handy on android smartphone

splaisan, 2024-11-04:
### Bambu Studio Version

2.15.1 (android)

### Where is the application from?

Bambu Lab Official website

### OS version

android (several phones tested)

### Additional system information

I bought a printer for my team and everyone would be happy to print on it.
Since there is not yet a way to access a printer from multiple accounts (please add this feature), I created a shared account to allow my colleagues to use the A1 printer at the office (we all get mails to that shared email address).

My plan is to detach the printer from my personal account and attach it to that new account using bambu handy logged into the new account BUT...

Although I can login in my own account using handy and pilot the printer, logging out then trying to log into the new account fails with the error message:

"Your account has been detected in another device. To keep it secure , please update the app and login again."

* no other device is currently logged into the account
* the account credentials I use are OK and work for bambu studio on my desktop and the bambu studio access has been validated with security code sent to the shared email.
* the same error is obtained with the smartphone of a colleague freshly installed handy but never used it before
* if I give my credentials in handy on my phone, I can login again so my app is functional with my own account.

can you please explain what is going on here, we are not even yet at the stage of attaching the printer to the app, we just cannot enter Handy at all.

thanks in advance

### Printer

A1

### How to reproduce

install handy on a phone
create a new bambu account and activate it in bambu studio
try to login in bambu handy android and get the error message

### Actual results

login fails

### Expected results

login works

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

btcmechanic, 2024-11-04:
I have the same issue on Android Phone.

JScoff, 2024-11-04:
Same issue on Android

leofidus, 2024-11-04:
This issue is preventing me from setting up my printer. I made an account when ordering, but when trying to log in on the phone I get the message "Your account has been detected in another device. To keep it secure , please update the app and login again.". I also get an email with a verification number, but the app doesn't offer a way to enter it. Without being able to log in I can't pair the printer.

mncasper78, 2024-11-05:
same issue.

MiDiANx360, 2024-11-05:
Happend to me last night too. Works now. I was just able to log into the app on my android phone.

splaisan, 2024-11-05:
apparently they fixed their error(s). Would have been nice to notify us though !!!
I finally got to the send code page and after filling it I could also access the app.
I let the dev team close this issue if it is really fixed at their side

leofidus, 2024-11-05:
Yeah, the app update today fixed it. Who knows, maybe it was already in the pipeline and just stuck in Google's review process a bit longer than anticipated.