bambulab/BambuStudio #5144: Ubuntu 20.04

belthaZornv, 2024-11-04:
### Bambu Studio Version

latest

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 20.04

### Printer

Bambu X1C

### How to reproduce

Installing via cmake

### Actual results

lto1: fatal error: bytecode stream in file ‘/home/julian/work/projects/BambuStudio_dep/usr/local/lib/libtbbmalloc.a’ generated with LTO version 6.2 instead of the expected 8.1
compilation terminated.
lto-wrapper: fatal error: /usr/bin/c++ returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/BambuStudio.dir/build.make:210: src/bambu-studio] Error 1
make[1]: *** [CMakeFiles/Makefile2:542: src/CMakeFiles/BambuStudio.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

### Expected results

installs

### Project file & Debug log uploads

None

### Checklist of files to include

- [X] Log file
- [X] Project file

belthaZornv, 2024-11-05:
Managed to get past that issue (with a ton of work!!), but now facing an issue:

```
/usr/bin/ld: /home/julian/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_core-3.1.a(imagpng.cpp.o): in function `(anonymous namespace)::wxPNGImageData::DoLoadPNGFile(wxImage*, (anonymous namespace)::wxPNGInfoStruct&) [clone .constprop.0]':
imagpng.cpp:(.text+0x3db7): undefined reference to `prusaslicer_png_create_read_struct'
/usr/bin/ld: imagpng.cpp:(.text+0x3dd2): undefined reference to `prusaslicer_png_set_read_fn'
/usr/bin/ld: imagpng.cpp:(.text+0x3ddb): undefined reference to `prusaslicer_png_create_info_struct'
/usr/bin/ld: imagpng.cpp:(.text+0x3e3d): undefined reference to `prusaslicer_png_read_info'
/usr/bin/ld: imagpng.cpp:(.text+0x3e68): undefined reference to `prusaslicer_png_get_IHDR'
/usr/bin/ld: imagpng.cpp:(.text+0x3e75): undefined reference to `prusaslicer_png_set_expand'
/usr/bin/ld: imagpng.cpp:(.text+0x3e7e): undefined reference to `prusaslicer_png_set_gray_to_rgb'
/usr/bin/ld: imagpng.cpp:(.text+0x3e87): undefined reference to `prusaslicer_png_set_strip_16'
/usr/bin/ld: imagpng.cpp:(.text+0x3e90): undefined reference to `prusaslicer_png_set_packing'
/usr/bin/ld: imagpng.cpp:(.text+0x3ed6): undefined reference to `prusaslicer_png_get_valid'
/usr/bin/ld: imagpng.cpp:(.text+0x3f68): undefined reference to `prusaslicer_png_read_image'
/usr/bin/ld: imagpng.cpp:(.text+0x3f75): undefined reference to `prusaslicer_png_read_end'
/usr/bin/ld: imagpng.cpp:(.text+0x3f9c): undefined reference to `prusaslicer_png_get_pHYs'
/usr/bin/ld: imagpng.cpp:(.text+0x404c): undefined reference to `prusaslicer_png_get_PLTE'
/usr/bin/ld: /home/julian/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_core-3.1.a(imagpng.cpp.o): in function `wxPNGHandler::LoadFile(wxImage*, wxInputStream&, bool, int)':
imagpng.cpp:(.text+0x5195): undefined reference to `prusaslicer_png_destroy_read_struct'
/usr/bin/ld: imagpng.cpp:(.text+0x51d2): undefined reference to `prusaslicer_png_destroy_read_struct'
/usr/bin/ld: /home/julian/work/projects/BambuStudio_dep/usr/local/lib/libwx_gtk3u_core-3.1.a(imagpng.cpp.o): in function `wxPNGHandler::LoadFile(wxImage*, wxInputStream&, bool, int) [clone .cold]':
imagpng.cpp:(.text.unlikely+0x613): undefined reference to `prusaslicer_png_destroy_read_struct'
/usr/bin/ld: imagpng.cpp:(.text.unlikely+0x697): undefined reference to `prusaslicer_png_destroy_read_struct'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/BambuStudio.dir/build.make:210: src/bambu-studio] Error 1
make[1]: *** [CMakeFiles/Makefile2:542: src/CMakeFiles/BambuStudio.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

MackBambu, 2024-11-05:
The code has undergone build testing before release.
https://github.com/bambulab/BambuStudio/actions/runs/11554318750
It is recommended to start by using BuildLinux.sh to build and test if everything functions correctly.
```
sudo ./BuildLinux.sh -u
./BuildLinux.sh -disr
```