bambulab/BambuStudio #5145: Cannot build on Fedora 39, Fedora 40, and Debian 12

smcgu, 2024-11-04:
### Bambu Studio Version

v01.09.07.52

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 39, Fedora 40, Debian 12

### Printer

X1C

### How to reproduce

1. Clone repository
2. Checkout `v01.09.07.52`
3. Install dependencies
4. Make `BambuStudio_dep` directory
5. Change into `BambuStudio/deps`
6. `cmake ../ -DDESTDIR="/home/user/Downloads/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1` (Step 2 of the guide, [building the deps](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide#2-building-the-deps))

The [Linux Compile Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide) dependencies did not seem to be complete for my Fedora and Debian installs. So, I'm noting them here. I understand that Debian is not officially supported but I was encountering the same error for two versions of (supported) Fedora.

- Fedora dependencies: `mesa-libGL mesa-libGLU mesa-libOSMesa nasm yasm x264 gstreamer1 gcc gcc-c++ cmake mesa-libGL-devel mesa-libGLU-devel m4 perl extra-cmake-modules wayland-devel wayland-protocols-devel libxkbcommon-devel gtk3-devel webkit2gtk3-devel gstreamer1-devel gstreamer1-plugins-base-devel gstreamer1-plugin-openh264 mesa-libOSMesa-devel nasm yasm x264-devel boost boost-devel libcurl-devel glfw-devel cereal-devel ilmbase-devel tbb tbb-devel glew-devel NLopt-devel openvdb-devel openvdb eigen3-devel expat-devel`
- Debian 12 dependencies: `cmake clang git g++ build-essential libgl1-mesa-dev m4 libwayland-dev libxkbcommon-dev wayland-protocols extra-cmake-modules pkgconf libglu1-mesa-dev libcairo2-dev libgtk-3-dev libsoup2.4-dev libwebkit2gtk-4.0-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad libosmesa6-dev nasm yasm libx264-dev libboost-dev libboost-filesystem-dev libboost-thread-dev libboost-log-dev libboost-locale-dev libboost-iostreams-dev libtbb-dev libcurlpp-dev libglew-dev libglfw3-dev libcereal-dev libnlopt-dev libnlopt-cxx-dev libopenvdb-tools`

Related issues and PRs:

- https://github.com/bambulab/BambuStudio/issues/2487
- https://github.com/bambulab/BambuStudio/issues/2606
- https://github.com/bambulab/BambuStudio/pull/5102

### Actual results

- [cmake_f39.txt](https://github.com/user-attachments/files/17624637/cmake_f39.txt)
- [cmake_d12.txt](https://github.com/user-attachments/files/17624636/cmake_d12.txt)

From Fedora 40:

```
-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- PROJECT_SOURCE_DIR: /home/user/Downloads/BambuStudio
-- CMAKE_MODULE_PATH: /home/user/Downloads/BambuStudio/cmake/modules/
-- Found PkgConfig: /usr/bin/pkg-config (found version "2.1.1") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found DBus: /usr/include/dbus-1.0;/usr/lib64/dbus-1.0/include  
-- LIBDIR: /home/user/Downloads/BambuStudio/src
-- LIBDIR_BIN: /home/user/Downloads/BambuStudio/deps/src
-- Found Boost: /usr/lib64/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams 
-- Boost::boost exists
-- TBB found in /usr/lib64/cmake/TBB
-- Found OpenSSL: /usr/lib64/libcrypto.so (found version "3.2.2")  
-- Found CURL: /usr/lib64/libcurl.so (found version "8.6.0") 
-- Found ZLIB: /usr/lib64/libz.so (found version "1.3.1")  
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3") 
-- Found EXPAT: /usr/lib64/libexpat.so (found version "2.6.3") 
-- Found PNG: /usr/lib64/libpng.so (found version "1.6.40") 
-- Found OpenGL: /usr/lib64/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.2.0") 
-- Found NLopt in '/usr/lib64'.
-- Using NLopt include directory '/usr/include'.
-- Using NLopt library '/usr/lib64/libnlopt.so'.
CMake Deprecation Warning at cmake/modules/FindOpenVDB.cmake:126 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  CMakeLists.txt:557 (find_package)

-- Found OpenVDB: /usr/include (found suitable version "11.0.0", minimum required is "5.0") found components: openvdb 
-- OpenVDB ABI Version: 11
CMake Warning at cmake/modules/FindOpenVDB.cmake:346 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:368 (just_fail)
  CMakeLists.txt:557 (find_package)

CMake Error at CMakeLists.txt:562 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

-- Configuring incomplete, errors occurred!
```

### Expected results

`cmake` should successfully configure.

### Project file & Debug log uploads

Only the Debian 12 build generates a `CMakeOutput.log` file. The Fedora `cmake` runs do not generate the log files.

- [CMakeOutput-debian_12.log](https://github.com/user-attachments/files/17624653/CMakeOutput-debian_12.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

MackBambu, 2024-11-05:
It is recommended to start by using BuildLinux.sh to build and test if everything functions correctly.
```
sudo ./BuildLinux.sh -u
./BuildLinux.sh -disr
```

smcgu, 2024-11-06:
This worked. Why is the `BuildLinux.sh` script listed as Step 4 of the [Linux Compile Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)? If it is the recommended (and working) solution, can it be bumped up to the top and recommended solution?

> It is recommended to start by using BuildLinux.sh to build and test if everything functions correctly.
> 
> ```
> sudo ./BuildLinux.sh -u
> ./BuildLinux.sh -disr
> ```

MackBambu, 2024-11-06:
> This worked. Why is the `BuildLinux.sh` script listed as Step 4 of the [Linux Compile Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)? If it is the recommended (and working) solution, can it be bumped up to the top and recommended solution?

The Linux compile guide hasn’t been updated in a long time, it may need to be rewritten.

mxmlnkn, 2025-06-17:
> This worked. Why is the `BuildLinux.sh` script listed as Step 4 of the [Linux Compile Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)? If it is the recommended (and working) [...]

I agree whole-heartedly! Please update the wiki.