bambulab/BambuStudio #5146: Additional part bug

ProduzioneMobilferro, 2024-11-05:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1 Carbon

### How to reproduce

- Loaded .stp file
- Cut object into 3 parts with 3 connectors for each cut
- Added negative parts, no problem, works just fine
- Added (normal) parts, specifically a cube, the software won't let me resize them. If I deselect and select again the additional part from the menu at the left, the software selects the connectors instead.

### Actual results

[image]

### Expected results

I should be able to add a part and resize it.

### Project file & Debug log uploads

[LAMA_R70 v5 modifiche.zip](https://github.com/user-attachments/files/17633684/LAMA_R70.v5.modifiche.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-11-06:
Thank you for your discovery, we will resolve it~

Haidiye00, 2024-11-06:
At present, you can first add an object, scale it, then select two, and merge them.

ProduzioneMobilferro, 2024-11-06:
> At present, you can first add an object, scale it, then select two, and merge them.

Bambu studio won't let me merge the cut section with another object. I think I could do this if it wasn't a cut section of an object. As it is, I haven't found a way to do what I need to do. The only thing I can do is to superimpose the new cube to the cut section, but the software gives me an error saying the 2 objects are in conflict. Since this is a big object I need to print in PAHT-CF, I don't want to try it only to have an error during print.

ProduzioneMobilferro, 2024-11-06:
[image]
As you can see, there is no "merge" command.

ProduzioneMobilferro, 2024-11-06:
For reference:
- Object cut in 3 parts with connectors
- Printed first part, found out that the part highlithed in the image is too small
- Before printing second part, I want to increase dimensions by merging a cube to the cut object
- I can't start from the whole object from scratch because there is no way I can place the connectors exactly as they were before. Starting from scratch would mean throwing away the first part I printed (about 800 g of PAHT-CF).
- Dosen't matter if first part is underdimensioned, as long as I can print 2nd and 3rd parts dimensioned correctly.

Haidiye00, 2024-11-06:
i see. Currently, you can add negative parts first, then scale it and right-click to modify the type, and select 'part'
[image]