bambulab/BambuStudio #5147: Brightness adjustment of the X1C camera

EEPHH, 2024-11-05:
Printing a dark material causes that you can't see anything on the camera. 
A glassriser with a bright LED stripe doesn't change anything, because the camera makes an auto adjustment, but far too dark.
Only thing you can do is, to change resolution, it will be a bit brighter, but you still can't see enough.
A brightness adjustment slider on the camera settings would be really really nice!

Hoot0118, 2024-12-29:
Please add a feature to let us control the auto exposure it dims it way to dark

Notahacker-420, 2025-01-19:
Yea I agree, I just installed a led riser and used an expensive 5730 led strip, brightest led strip i've seen, and yet, somehow, the live feed ignores the extra light. I specifically did the riser upgrade to fix this issue, thinking it was the very small led light installed in my X1C, I was wrong. Now it seems like it's over-compensating for the bright light and it's even dimmer than before.

lauretiu, 2025-03-13:
It's i curating the way the camera handles the exposure. If you install a riser it looks the same or worst in terms of brightness seen through the camera. We need a manual slider. Please give us that option!

camusyan, 2025-09-14:
Same here. Camera is just useless when printing black material. Exposure is way too dark and we can't see anything...