bambulab/BambuStudio #5150: Inner wall is printed outside of the model

pesk, 2024-11-06:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1. Open attachted project file
2. see layer 16

### Actual results

The inner wall crosses the outer walls and prints where it should not be printed. This happens with classic wall mode:
[image]
[image]

### Expected results

The inner wall sould not be printed outside.
[image]

Switching to arachne wall mode produces the expected result.

### Project file & Debug log uploads

[bug.zip](https://github.com/user-attachments/files/17644834/bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

pesk, 2024-11-06:
Sorry, uploaded the file bug.zip with arachne wall mode.
See the attached file with the classic wall mode where problem occurs.
[bug-classic.zip](https://github.com/user-attachments/files/17644967/bug-classic.zip)

ttt888zzz, 2024-11-11:
HI [pesk](https://github.com/pesk)：
Please download the latest version and visit the website，https://bambulab.cn/zh-cn/download/studio

closed 2025-04-03 by QingZhangBambu