bambulab/BambuStudio #5155: a1 mini camera and light inop

nevermore79, 2024-11-06:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1

### Additional system information

Apple M!
16 gb

### Printer

ai mini

### How to reproduce

Open up application printer a1 mini an over near selecting to play the camera it shows camera malfunction
If you open up Control on the actual printer and select the light button you can see that it lights up around the light and the light never turns on on the machine
This also happens in the bamboo app on the iPhone when selecting the printer a one mini and question it does not allow you to see a a notification to play video or view it just shows a blank screen where you should be able to access the video footage or live footage and clicking on the light does not do anything
I have deleted and reinstalled the application to see if that fixed it
It did not
I have turned off a one mini printer and turned back on to see if that would fix the light not working or the camera not being able to view the build plate

### Actual results

Nothing happens no light no camera action

https://github.com/user-attachments/assets/8d0cef7f-4f38-46b1-9b33-363aeb7f25f3
[image]

### Expected results

Anytime you're in the app and you click view you should be seeing what is being done on the printer so if you're printing something out on the printer you should see the print bed in the item being printed if the printer is at idle and nothing is being printed you should see just the bed. Or if the item has finished printing and you view the camera you should see you're finished print.
The same pretty much goes for the light if you click the button on the light should be showing that it's on and lit up but anytime I click on the light in the application on the iPhone it shows the light is on but if you look at the A1 mini printer the light is not on
If you turn the light on from the touchscreen on the a1 mini the light shows that it's on for a brief moment and then turns itself off I believe that's due to the fact the light is not working
When in bamboo Studios on PC or Mac if you turn the light on for the a1 mini through the application it stays on for a moment or two and then turns itself off I may believe that this is caused by the fact that the light does not work

### Project file & Debug log uploads

We don't support that file type

### Checklist of files to include

- [X] Log file
- [X] Project file

smashedllama, 2024-11-08:
I am having the same issue on two of my 3 A1 mini printers. The two with AMS units do not work but the one with no ams works fine. All on the same latest FW with the ams and the printer. One of mine has the light issue and the other does not. 
[image]

nevermore79, 2024-11-08:
dose the one with a non working light also have a camera problem

nevermore79, 2024-11-08:
dose the mini with the light issue also have a non-working camera ?

closed 2024-11-08 by nevermore79

smashedllama, 2024-11-08:
both of the units with the ams have the issue. One has the light issue and one does not.

jonathanoeijoeng, 2025-06-28:
is there any update on this? I just got same situation just now