bambulab/BambuStudio #5165: AMS drops filament presets when Studio opens

AlAmantea, 2024-11-09:
### Bambu Studio Version

1.10.0.81

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

X1 Carbon

### How to reproduce

Set filament vendor, color, and material in the printer for the AMS unit.
Verify all 4 slots are loaded with filament and ready for use.
Open Studio
Select the correct printer in Prepare tab
Go to device tab
AMS filament settings will clear themselves randomly, especially on slots 1 and 2

### Actual results

Set filament vendor, color, and material in the printer for the AMS unit.
Verify all 4 slots are loaded with filament and ready for use.
Open Studio
Select the correct printer in Prepare tab
Go to device tab
AMS filament settings will clear themselves randomly, especially on slots 1 and 2

### Expected results

Set filament vendor, color, and material in the printer for the AMS unit.
Verify all 4 slots are loaded with filament and ready for use.
Open Studio
Select the correct printer in Prepare tab
Go to device tab
AMS filament settings should populate with the correct filament as chosen on the printer and not randomly disappear.

### Project file & Debug log uploads

no uploads need to reproduce this bug.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-12-02:
@AlAmantea  Can your reproduce this problem? And can you offer your X1C firmware version for us to get a further test?

AlAmantea, 2024-12-06:
@JohnSmithCYM Yes, the problem is reproduceable, although it has gotten a little better with the latest studio update (1.10.1.50).

I am currently running the latest version of firmware (1.08.02.00)