bambulab/BambuStudio #5166: Missing preview on Prepare and Preview panes

petarkolaric, 2024-11-09:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Bazzite 40 (Based on Fedora SilverBlue)

### Additional system information

CPU: AMD Ryzen™ 7 7800X3D × 16
Video: Nvidia 4080 Super 16gb
Motherboard: Gigabyte B650M

### Printer

Bambu P1S

### How to reproduce

1. Open Application
2. Find model in online section in home (I chose the V twin Bambu 005 project)
3. View preview for model

### Actual results

Preview is missing
Plate information colour scheme is very difficult to read
[image]

### Expected results

Preview should display
Plate information text should have enough contrast to easily be read

### Project file & Debug log uploads

[Engine Model Components Kit-005 beta 5.zip](https://github.com/user-attachments/files/17685928/Engine.Model.Components.Kit-005.beta.5.zip)
[debug_Sat_Nov_09_14_4.zip](https://github.com/user-attachments/files/17685947/debug_Sat_Nov_09_14_4.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

vkukk, 2024-11-10:
Exactly the same issue on Fedora 41 with both AppImage downloaded from Github or installed via Gnome Software Flatpak.

Nvidia and Wayland.

MackBambu, 2024-11-11:
This is a known issue; try downgrading the driver to version 555.
https://github.com/bambulab/BambuStudio/issues/4669

vkukk, 2024-11-11:
> This is a known issue; try downgrading the driver to version 555. #4669

No.
Blaming nvidia drivers when all the other software works fine is not solution. I will not downgrade because meanwhile there have been great improvements in driver. You must fix your broken software, not to demand people downgrading their drivers.

mabl, 2024-11-11:
This may be related to https://github.com/bambulab/BambuStudio/pull/923. I might be wrong, but I believe EGL is the main backend on Wayland.

MackBambu, 2024-11-12:
> No. Blaming nvidia drivers when all the other software works fine is not solution. I will not downgrade because meanwhile there have been great improvements in driver. You must fix your broken [...]

The work is already underway, and we are currently considering whether the issue is due to a lower version of OpenGL. However, we are also puzzled by NVIDIA dropping support for lower versions of OpenGL.

Arcau, 2024-11-15:
I am also having this same issue on Fedora 40 - I have manually set the flatpak to display with x11 and still the same 
version 1.10.0 public release 
[image]

fireshaper, 2024-11-23:
A comment on issue https://github.com/bambulab/BambuStudio/issues/4626 mention adding this to the environment variables of the appimage:

`__GLX_VENDOR_LIBRARY_NAME=mesa  __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json`

This worked for me.

Arcau, 2024-11-24:
> I am also having this same issue on Fedora 40 - I have manually set the flatpak to display with x11 and still the same version 1.10.0 public release [...]

This will ONLY work with the appimage version of the software and NOT the flatpak - but i does render with those args

killua99, 2024-12-06:
In the meantime, what tool could we use to make it work, or to be able to print on Linux?

jarrard, 2024-12-14:
its very broken, at least for nvidia users.

NAM1025, 2024-12-20:
Same issue with AppImage and Flatpak. Using NVIDIA drivers 565.77

Adding the env variables works but it is incredibly laggy with multi-color prints to the point that it's not even use-able. I've had to swap back to Windows in order to use it.

jarrard, 2024-12-20:
Yep its VERY slow using NVIDIA card.  Very unfortunate.

powermad80, 2024-12-28:
Switching to x11 mode in flatseal does not resolve this in my install of Fedora 41 KDE with Nvidia. In fact, it was already set to that by default. Changing any settings only causes it to not launch at all.

NAM1025, 2024-12-28:
I will say, swapping to KDE Plasma X11 (instead of wayland) and removing the environment variables to use the mesa drivers, does make it use-able again... but I'd prefer to stay on Wayland.

LordyPandaz, 2025-03-17:
Considering the current Fedora kernel (6.13.6-200.fc41.x86_64) does not work with older Nvidia drivers now (570.124.04 does), this really needs to be fixed.

vkukk, 2025-03-19:
Things like this prevent people from using Linux on desktop. Just imagine Bambu Studio would be broken on MS Windows for half a year like it has been for Linux.

Shame on you, Bambulab!

jarrard, 2025-03-19:
Well a lot of developers are struggling to adapt to Wayland since this issue appears to not happen under X11.  

Also its mostly fixed in OrcaSlicer now.  Still needs 1 work around for klipper printers.

NAM1025, 2025-03-20:
For OrcaSlicer, using the zink mesa drivers fixed these issues for me. I'm assuming they'll work for BambuSlicer as well.

Add these environment variables
`__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink WEBKIT_DISABLE_DMABUF_RENDERER=1`

haenno, 2025-07-01:
Had the same issue, nearly same hardware but on Arch. 
Solution found here (with Flatpack instead of AUR): https://github.com/bambulab/BambuStudio/issues/4669#issuecomment-2923389879

chiefy, 2026-03-23:
@NAM1025 thank you! Had same issue on CachyOS and your solution works.