bambulab/BambuStudio #5167: Search function for files doesn't work on mac

NEMES1SV12, 2024-11-09:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

masOS 15.1

### Additional system information

16" macbook pro, M1 pro 16gb ram

### Printer

Bambu Lab X1C, P1S

### How to reproduce

When I go to add an object to the print bed, the window opens to find the file, when I try to use the search function, it doesn't ever give results.  If I go to finder on my mac, it works just like it should.

### Actual results

I'm not allowed to use the search function

### Expected results

Be able to see a relevant result in the pop up window to add a file to the print project

### Project file & Debug log uploads

I don't have one

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-11-11:
HI [NEMES1SV12](https://github.com/NEMES1SV12)：Could you please Provide a video to help check the issue，thanks.
[image]

NEMES1SV12, 2024-11-11:
In the video I’ve opened bambu studio and I’m trying to add a file to the print plate.  after the window opens up you can see the file I’m trying to use is already there but when I type it in the search, no files pop up

image.png (view on web)

Haidiye00, 2024-11-12:
Hello, can you download ScreenToGif software and record a GIF to upload? Thank you~

Haidiye00, 2024-11-12:
[image]
Are you referring to this search button. This button can only search for preset related things.

NEMES1SV12, 2024-11-13:
When I open bambu studio and try to add a file to my print, I click the button in the tool bar above the print bed and a window opens to add files.  When I use the search in the top right of that window, no results show up.  I have even typed in the file names that I can see when the window opens and the search still won’t show any results.  When I use finder on my MacBook it finds the file in under a second and I can open it, but for some reason trying to search in the window that opens up in bambu studio, it doesn’t work at all.

Haidiye00, 2024-11-15:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you