bambulab/BambuStudio #5171: Problems building Bambu Studio

adaviel, 2024-11-09:
### Bambu Studio Version

01.09.07.52

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 41 6.11.5-300.fc41.x86_64

### Printer

A1 mini

### How to reproduce

$ unzip BambuStudio-01.09.07.52.zip
$ cd BambuStudio-01.09.07.52
optionally edit BuildLinux.sh to set BUILD_ARGS="-DDEP_WX_GTK3=ON -DJPEG_VERSION=6"
$ ./BuildLinux.sh -dr |tee BuildLinux2.log

### Actual results

[135/180] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/andrew/bambu/BambuStudio-01.09.07.52/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 
cd /home/andrew/bambu/BambuStudio-01.09.07.52/deps/build/dep_OCCT-prefix/src/dep_OCCT && /usr/bin/git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /home/andrew/bambu/BambuStudio-01.09.07.52/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /home/andrew/bambu/BambuStudio-01.09.07.52/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
ninja: build stopped: subcommand failed.

### Expected results

I expect to get a working BambuStudio.

I initially tried to build on my laptop running CentOS 8, but had various issues. I then installed Fedora 41 on an old laptop, which is supported in BuildLinux.sh. I have tried releases 01.09.07.52, 01.10.00.81, and the master on Github. I have tried BuildLinux.sh and also cmake per the Linux Compile Guide. I have tried cloning using Git, but since I'm in a rural areas without cable internet that doesn't work too well (e.g. getting wxWidgets) so I've downloaded zipfiles.

On one occasion I was able to finish building the dependencies on CentOS 8.When I tried building Bambu Studio I got
$ cd /home/andrew/bambu/BambuStudio-01.10.00.81
$ mkdir install_dir
$ mkdir build;cd build
$ cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="~/bambu/BambuStudio81_dep/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release |tee cmake.log

--   Package 'libavcodec', required by 'virtual:world', not found
--   Package 'libswscale', required by 'virtual:world', not found
--   Package 'libavutil', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:607 (message):
  A required package was not found

I had hoped that Buildlinux on Fedora would be more reliable. I've been building applications occasionally on Linux for 30 years (which means that 95% of what I know is now wrong..). This has been particularly frustrating - tantalizingly close to working, but large and complex enough that I can't easily find the problems and fix them.

### Project file & Debug log uploads

From BuildLinux on Fedora 41
[BuildLinux2.log](https://github.com/user-attachments/files/17689441/BuildLinux2.log)

From cmake on CentOS 8
[cmake.log](https://github.com/user-attachments/files/17689440/cmake.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-11-11:
The ffmpeg build was not successful. Did you run "sudo ./BuildLinux.sh -u" to ensure the dependencies were successfully installed?

adaviel, 2024-11-17:
I did run "./BuildLinux.sh -u" on Fedora 41.

All the dependency RPMs in linux.d/fedora seem to be installed, mostly from Fedora but x264-devel from RPM Fusion.

BuildLinux.sh -d  on Fedora 41 still hangs with
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/andrew/bambu/BambuStudio-master/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 

Incidentally, in linux.d/fedora there's a line if [ $fedora_version == "40" ], so I have webkit2gtk4.0-devel. webkit2gtk4.1-devel is available from the Fedora 41 repository; should I install that ?

=============
BuildLinux.sh won't run on Centos8
With cmake I got 
Package 'libavcodec', required by 'virtual:world', not found
I can see e.g /home/andrew/bambu/BambuStudio81_dep/usr/local/lib/libavcodec.so.61.3.

I also have ffmpeg-4.2.9-1.el8.x86_64  previously installed to play videos. That has e.g. /usr/lib64/libavcodec.so.58.54.100
If install  ffmpeg-devel with dnf, I am able to get past that error and complete the first cmake

When I try the second cmake "cmake --build . --target install --config Release -j1", I get
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:204: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Widgets/ProgressDialog.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1139: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2

If I try "make -i" to see what else there might be, I get a lot more errors - for instance a lot of "‘wxWebViewEvent’ has not been declared, and a complaint about braces in StateHandler.cpp that's easy to fix.

================
I am going to install Fedora 40 on the old laptop and try again with BuildLinux.sh.

MackBambu, 2024-11-24:
> In chroot, I set DISPLAY=:0 and xev and bambu-studio run, except the 3D viewer does not work. I get MESA: error: Failed to query drm device. libEGL warning: egl: failed to create dri2 screen error: XDG_RUNTIME_DIR is invalid or not set in the environment. MESA: error: ZINK: failed to choose pdev libEGL warning: egl: failed to create dri2 screen Is it possible to fix this ? My centos8 laptop is faster with a bigger screen than the old one I put FC40 on.

The issue of not displaying the 3D interface is a known problem. We currently suspect it is related to the OpenGL version and are working on a fix.
You can refer to the methods in this post, such as downgrading the GPU driver or switching to  #X11. #4669

adaviel, 2024-12-03:
I tried 
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 /bambu/BambuStudio/bin/bambu-studio  |& tee bambustudio.2.log

I still get
MESA: error: Failed to query drm device.
libEGL warning: egl: failed to create dri2 screen
MESA: error: ZINK: failed to choose pdev

Out of interest, I triedd running glxgears in the chroot environment. I get
MESA: error: Failed to query drm device.
glx: failed to create dri3 screen
failed to load driver: iris
but the application still runs and produces graphics.

I realize this is an edge case (running FC40 chroot on Centos8) I just wondered if there was something I was doing stupid that had a simple fix.

adaviel, 2024-12-16:
Thank you; that appears to work. I'd not heard of flatpak, but it seems to be installed on my Centos8 system. Video appears to work too.

If I'd known about that, I'd have tried it to start with. But there seemed to be only Mac and Windows pre-built applications listed, no RPM  or DEB packages (or Flatapp, or Appimage, or Docker etc,)

MackBambu, 2024-12-23:
> If I'd known about that, I'd have tried it to start with. But there seemed to be only Mac and Windows pre-built applications listed, no RPM or DEB packages (or Flatapp, or Appimage, or Docker etc,)

The Flatpak version is ported by third-party volunteers. While we occasionally provide assistance, it is not hosted on the official website. However, it can serve as an alternative.

closed 2025-04-29 by MackBambu