bambulab/BambuStudio #5172: Camera stream stops working after one second

TheElbinho, 2024-11-09:
### Bambu Studio Version

1.9.7.52 (Flatpak)

### Where is the application from?

Bambu Lab github releases

### OS version

Universal Blue aurora-surface

### Additional system information

Surface Pro 5

### Printer

X1C

### How to reproduce

1. Open Bambu Studio
2. Go to device tab
3. Add printer (LAN mode)
4. Start Camera Stream

### Actual results

Video is displayed for 1 second, then freezes

### Expected results

continuous streaming

### Project file & Debug log uploads

This is the GStreamer log output. I don't know how to process any of this, because GStreamer will always be black magic to me
[bambulog.txt](https://github.com/user-attachments/files/17689542/bambulog.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-11-11:
Can you help try the latest beta version 1.10?

bambu123, 2024-11-11:
> I don't think the beta versions are available as flatpaks, or are they?

Not yet.

AppImage?
https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.81/Bambu_Studio_linux_fedora-v01.10.00.81.AppImage

TheElbinho, 2024-11-18:
I can in fact confirm that the camera stream just works in version 1.10.1.5

closed 2024-11-18 by TheElbinho