bambulab/BambuStudio #5173: 0.8 mm nozel layer height not working

woolmonkey, 2024-11-10:
### Bambu Studio Version

1.9.7.25

### Where is the application from?

Bambu Lab Official website

### OS version

Linux

### Additional system information

Flatpak

### Printer

Bambu X1C

### How to reproduce

Change layer height to 0.24

### Actual results

Layer height doesn't change stays at 0.4

### Expected results

Expect layer height to change to 0.24

### Project file & Debug log uploads

Not for printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-11-11:
Please provide the project and log files.

woolmonkey, 2024-11-11:
Project and log files from what?  I just noticed this happening in the slicer when changing the layer size with a 0.8 mm nozzle.  Doesn't happen with 0.4 mm nozzle.

closed 2024-11-11 by woolmonkey