bambulab/BambuStudio #5174: Change the filament for a whole plate at once

cuban2, 2024-11-10:
**Is your feature request related to a problem? Please describe.**
when I am doing a multi-color print and separate my plates by color I have to go to each object on that plate to change the filament when I want to change color.

**Describe the solution you'd like**
It would be great that under Process -> Objects that the plate could have a filament selection option so that you could adjust the color or type of filament for a whole plate at once

[2024-11-10_12-47-16.pdf](https://github.com/user-attachments/files/17693020/2024-11-10_12-47-16.pdf)

xSepticSidx, 2024-11-10:
Can you not just select all with Ctrl+A and then hit a filament number?

cuban2, 2024-11-11:
Yes I can. Thank you please disregard my suggestion	
Rudy Garcia
***@***.*** ***@***.***>
Cell: 909-239-6992