bambulab/BambuStudio #5177: Support Slicer Issue - Irregular/Jagged tree support with Archane slicer and `Initial Layer Expansion` increased

CraigHutchinson, 2024-11-11:
### Bambu Studio Version

1.8.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

1. Using Archane Slicer
2. Increase the 'Initial Layer Expansion'
3. Slice model and observe the Tree support slices
4. Repeat step 2>3 and supports incrementally degrade!

### Actual results

After changing the expansion a couple of times from 2..3..4 we see the support tree is 'jagged' and there are lots of irregular slicing lines which causes supports to fail mid-print as far as I can tell. i.e. The layers are not stacked ontop of each other due to this issue!
[image]
[image]

### Expected results

Issue can be resolved by toggling back to 'Classic'<>`Archane` and issue corrects/goes-away...
[image]
[image]

### Project file & Debug log uploads

[2024_11_11_TreeSupportSlice_Issue_Testa di Drago_B.zip](https://github.com/user-attachments/files/17700392/2024_11_11_TreeSupportSlice_Issue_Testa.di.Drago_B.zip)
[2024-11-11 log.zip](https://github.com/user-attachments/files/17700434/2024-11-11.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-11-12:
Please wait for the next new release and see if the problem exists.

tangketan, 2024-11-13:
@CraigHutchinson Hi, the new version has been released. Did you try it?

CraigHutchinson, 2024-11-13:
There is very little - almost nothing - of 'Support Interface' being generated for supports but I don't know if thsta a bug or it isn't expected? I can open a new bug if you see this as one.

I'm happy to say the jagged support  issue is resolved.

CraigHutchinson, 2024-11-15:
I would confirm this appears fixed in v1.10 and I have not seen this issue since.

closed 2024-11-15 by CraigHutchinson