bambulab/BambuStudio #5183: Inability to print supports

WoodFishpond, 2024-11-12:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: i7-11700, 16 GB DDR4 3200mt, geforce rtx 3060

### Printer

Bambu Lab A1

### How to reproduce

1. Open Bambu Studio
2. Slice file that needs support. Pin tests also work.
3. Send to print
4. Wait for supports to fail by being forcefully ripped off the bed by the print head.

### Actual results

I can get a video if you need it. Super easy. No matter how large of a brim I give the supports, the head of the printer pushes on them enough to rip them off and misprint.

### Expected results

I should have a perfectly fine model. My brother used his computer to slice it and it works just fine. I downloaded Studio on another computer and it still doesn't work.

### Project file & Debug log uploads

Anything will do

### Checklist of files to include

- [ ] Log file
- [ ] Project file

killsecurly, 2024-11-12:
Are you using a user preset or BBL's presets?

WoodFishpond, 2024-11-13:
Both their presets and default on basic stl files.

ttt888zzz, 2024-11-13:
HI [WoodFishpond](https://github.com/WoodFishpond)：Could you please Provide the 3mf file  and the video，thank you
[image]