bambulab/BambuStudio #5185: [BUG] manually applied rotation always show as 0

borsch, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

irrelevant

### How to reproduce

watch video

### Actual results

rotation degrees always show as 0

### Expected results

rotation degrees should be shown with value that was inserted into into input

### Project file & Debug log uploads

issue with 1.10.0.89 [bambu_studio_rotation_issue.webm](https://github.com/user-attachments/assets/4eded4bf-37cd-4bf6-8caa-8c18a6de2be4)

as it was was with 1.9.7 
[bambu_studio_1_edit_0.webm](https://github.com/user-attachments/assets/84320d21-43c3-4cbe-b640-705e69447947)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-01-06:
Hello, this issue has been fixed in our internal version~

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00