bambulab/BambuStudio #5186: Importing Step File fails

benjaminkott, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Import Step file
2. See dialog 
    [image]
3. Change parameters
4. See calculation never finishes
5. See Exception 
[image]

### Actual results

BambuStudio crashes due to unhandled exception

### Expected results

Model is imported as before

### Project file & Debug log uploads

[Step Files.zip](http
[logs.zip](https://github.com/user-attachments/files/17714473/logs.zip)
s://github.com/user-attachments/files/17714447/Step.Files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Hamudulu, 2024-11-12:
Exact the same issue here....

nmkr, 2024-11-12:
+1 same here

Laurent-K, 2024-11-12:
same for me

MackBambu, 2024-11-12:
Could you send me the configuration file? It’s located in C:\Users\USERNAME\AppData\Roaming\BambuStudio. After that, you can try clearing the configuration file and try again.

nmkr, 2024-11-12:
[BambuStudio.zip](https://github.com/user-attachments/files/17717037/BambuStudio.zip)

added the existing config as zip file

[image]

after deleting the configuration file and reopening bambustudio the error still exists

MackBambu, 2024-11-12:
Hey everyone, I found the cause, it’s due to an incorrect string conversion format. Please switch your locale to English for now, and I’ll fix the issue shortly.

nmkr, 2024-11-12:
> Hey everyone, I found the cause, it’s due to an incorrect string conversion format. Please switch your locale to English for now, and I’ll fix the issue shortly.

can confirm - switching from german to english resolved the issue

benjaminkott, 2024-11-12:
@MackBambu can confirm, can now be loaded.

[image]
[image]

Laurent-K, 2024-11-12:
HI, ok I did it and now it works but I had to close and open 2 times.

MartinChan3, 2024-11-12:
@Laurent-K Have you tried switch Language to English?

Laurent-K, 2024-11-12:
Yes and it works with English language. Sorry if my explaination was not ok. the first time I did it it crashed, but I opened the software again and it works now. Thank you !

Givaro72, 2024-11-12:
I tried English but no luck importing a stp file.
OBJ works but the only thing what i see is that circles ar not round anymore, but divided in segments. 
Where to change that ?
For now I downgraded to the previous version.

Steff1964, 2024-11-12:
Same here, will try English version or downgrade.

SorenTheSwede, 2024-11-12:
Can also confirm that switching from Swedish to English solved the problem

Vespisti1969, 2024-11-12:
But that cannot be the solution. There are people who do not speak or understand English.

benjaminkott, 2024-11-12:
@Vespisti1969 please see https://github.com/bambulab/BambuStudio/issues/5186#issuecomment-2470555626 it´s a workaround until a patch was issued.

Vespisti1969, 2024-11-12:
> @Vespisti1969 please see [#5186 (comment)](https://github.com/bambulab/BambuStudio/issues/5186#issuecomment-2470555626) it´s a workaround until a patch was issued.

I‘ve read it and use this workaround. For me its ok, but maybe other people have a problem to undrstand english well. Im sure that bambu will fix the issue very fast.

boinjoe, 2024-11-13:
For me it was not working. Switching from German to englisch made it Wiese. Now the program crashes directly after opening.

MackBambu, 2024-11-13:
> Please also add detailed help on the Step File Import Parameters dialog that explains what the impact is of the settings. Mouseover help (like a ?) and detailed help added to the Wiki pages would be helpful.

I will be updating the wiki soon.

MackBambu, 2024-11-13:
> > HI, ok I did it and now it works but I had to close and open 2 times.
> 
> Same here.

Clean up the configuration file or open BambuStudio.conf, find the lines for angle_deflection and linear_deflection, and delete them. They might have been set to incorrect values.

MackBambu, 2024-11-13:
> How about using simple, understandable terminology in the dialog like "Higher resolution/detail" and "Lower resolution/detail"? The numerical values can still be displayed and editable, but please it costs nothing to make it easier for users to understand.
> Thanks

Thank you, we will consider your suggestions.

fahaaaxiki, 2024-11-14:
**V1.10.0 Hotfix Preview**
**Windows**:https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
**Mac**:https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg
We have urgently fixed issues including STEP import crashes, lag on Mac(Intel CPU) after slicing, and temperature display errors.  Please use the above version to resolve these issues you may encounter before we officially release the Hotfix version.

MackBambu, 2024-11-15:
> If Cancel is selected, the import cancels immediately. But if either of the settings are moved, and then Cancel is selected, it takes a very long time (with a large CAD model) to Cancel while the number of triangles is calculated - it should also be immediate and the calculation of the number of triangles should be aborted/interrupted immediately. I have had to force close BS to restart (with a large CAD model).

In fact, I have optimized this part. Could you kindly share the STEP file?

MackBambu, 2024-11-18:
> @MackBambu Sure - but I cannot post it publicly and there's no PM system that I can find on Github - how can I email it to you privately, purely for the purposes of testing this and with the assurance that it will be deleted thereafter and not shared with anyone else?

You can contact the support team and submit a screenshot of our chat. They will discreetly forward the file to me.

MackBambu, 2024-11-20:
@3DWORX-SA 
I tested loading a file of around 100MB, and there was no issue with clicking cancel taking a long time.
https://drive.google.com/file/d/10e-Njzs9TOQcYCJvqd1gGaMnJ3Fndnsw/view

closed 2025-01-24 by benjaminkott