bambulab/BambuStudio #5189: cannot import STEP file anymore

Laurent-K, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 pro 64b 23H2

### Additional system information

13th Gen Intel(R) Core(TM) i9-13980HX   2.20 GHz
Ram 64GB
NVIDIA RTX 5000 Ada

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Before the intallation of the update today I was able to import STEP but now I can not import. As soon as I m trying I get this new windows about tolerences and the software crash every time ( I have tryied 5 times and always the same)

### Actual results

the software crash every time

### Expected results

the STEP file should be imported as usual and no crash

### Project file & Debug log uploads

I can not save anything because the software crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Laurent-K, 2024-11-12:
how can I go back to a previous version until you get that fixed ?

Laurent-K, 2024-11-12:
[image]

benjaminkott, 2024-11-12:
Same issue as https://github.com/bambulab/BambuStudio/issues/5186

Peter28357, 2024-11-12:
Same issue
[ZweiFarben - Part 1.zip](https://github.com/user-attachments/files/17716733/ZweiFarben.-.Part.1.zip)

MackBambu, 2024-11-12:
@Laurent-K @Peter28357 
The merging issue is discussed in #5186, thanks

S-Risak, 2024-11-12:
@Laurent-K 
> how can I go back to a previous version until you get that fixed ?

Download previous version at: https://github.com/bambulab/BambuStudio/releases/tag/v01.09.07.52 
Just install as normal update.

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50