bambulab/BambuStudio #5190: Finish print time not working correctly

Arnigon, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Verison 10.0.22631 Build 22631

### Additional system information

CPU	13th Gen Intel(R) Core(TM) i7-13700K
GPU: Gainward 4080 Panther
RAM: Corsair 32Gb DDR5 6400MHz

### Printer

Bambu Lab P1S

### How to reproduce

1. Start Bambu Studio
2. Import STL
3. Slice file with own settings
4. Send to printer

### Actual results

Finish print time text box is there, but not showing any actual time.
[image]

### Expected results

There should be a estimated finish time present

### Project file & Debug log uploads

[Project file.zip](https://github.com/user-attachments/files/17716529/Project.file.zip)
[debug_network_Tue_Nov_12_12_01_56.log.zip](https://github.com/user-attachments/files/17716576/debug_network_Tue_Nov_12_12_01_56.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Guenni1975, 2024-11-12:
Interestingly, the problem also occurs on my HP notebook with i7 processor.
The end time is displayed on my Huawei Mate Book with i5 processor.
So the problem does not seem to exist on all computers!

Tumsk, 2024-11-13:
The issue also happens to me. 
Finish Time is never populated with a value.
Printing from .3mf (not tried an imported .stl)

Bambu Studio: 1.10.0.89
Printer: P1S
CPU: Intel(R) Core(TM) i7-8700 CPU
RAM: 32.0 GB
OS: Win 11 Pro 24H2 
OS Build: 26100.2314

DanBao-Bambu, 2024-11-18:
Hello, the cause of this issue may be related to the screen zoom ratio. You can try adjusting the screen zoom ratio appropriately and fix the problem in the next version.