bambulab/BambuStudio #5191: X1C: failed to obtain cloud custom material information

stedebus, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. created custom Filament in Bambu Studio
2. selected that Filament for AMS Slot
3. after a few seconds the filament disapeared from that slot and "?" is displayed
4. Tried to select that Filament on the Printer Display, but it is not listed
5. click on "resync" symbol next to the material button
6. Error "failed to obtain cloud custom material information" shows up

### Actual results

Error "failed to obtain cloud custom material information" shows up

### Expected results

Printer should resync materials with cloud
[image]

### Project file & Debug log uploads

Problem is not bound to a project

### Checklist of files to include

- [X] Log file
- [X] Project file

MartinChan3, 2024-11-12:
I have tried it with my X1C and nothing wrong, could you tell me which version of X1C firmware do you use?

stedebus, 2024-11-12:
Hi,
Version 01.08.02.00 (latest)
It is not the first time that I use custom filament. There is already some stored in the cloud and printer had synced it. But as I tried today, I got that error. And I saw a custom Filament on the printer, which I have deleted some days ago in Studio SW.

Regards
Stefan

stedebus, 2024-11-12:
The problem could be solved by loging the printer out of my account, and back in again. After that, everything was synced correctly with the cloud.

closed 2024-11-12 by stedebus