bambulab/BambuStudio #5192: step import crashes the program

boinjoe, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 11

### Additional system information

CPU: Intel Core i9-9600
Nvidea GTX 2070

### Printer

Bambu P1S Combo

### How to reproduce

1. open new project
2. import step files
3. step file import parameter window opens
4. change paramaters
5. hitting OK for continue

### Actual results

error message is shown
"unhandeld unknown exception, terminating the application"
bambu studio crashes

### Expected results

import step

### Project file & Debug log uploads

[crash_Tue_Nov_12_12_41_52_0_done.zip](https://github.com/user-attachments/files/17717355/crash_Tue_Nov_12_12_41_52_0_done.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

MartinChan3, 2024-11-12:
You can switch your language to English to solve this problem.

boinjoe, 2024-11-12:
Thanks for support. Unfortunately the change made everything worse. No the error message appears directly after start of program.Best regards Jörg Ziegler

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg