bambulab/BambuStudio #5194: Bambu Studio very very slow on Mac Monterey

HO44-PROJECT, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 12.6.2

### Additional system information

2,2 GHz Quad-Core Intel Core i5
16 GB 1600 MHz DDR3

### Printer

Bambu Lab A1

### How to reproduce

Load 3mf, place object, Slice...

### Actual results

Wheel of death, application not responding, lags, unusable

### Expected results

Just like before :-)

### Project file & Debug log uploads

[Chirtmas+Elf+with+holes.3mf.zip](https://github.com/user-attachments/files/17718628/Chirtmas%2BElf%2Bwith%2Bholes.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

mikegabriel, 2024-11-12:
Same issue here after updating to v1.10.0 this morning.  In my case I think it's when I have more than 1 plate in my design file.

If I have all my models on a single plate and hit slice, everything is good.  If I add a new plate and drag 1 model to it then slice that plate, performance drops to like 1fps and I can no longer pan around the camera on the preview tab.

I'm on macOS Sonoma 14.6.1.

zanuda2raza, 2024-11-12:
Same macOS 15

mikegabriel, 2024-11-12:
For curiosity, is everyone on an Intel chipset or do we have issues across both Intel and M series? 

I'm on an Intel I9 2.3GHz.

HO44-PROJECT, 2024-11-12:
Intel here and multiple plates also

aayevtushenko, 2024-11-12:
same on Intel Mac OS

sagatxxx, 2024-11-12:
Same here, intel osx iMac 27 late 2013

northdakota, 2024-11-12:
Same thing Monterey M1 chip

lanewei120, 2024-11-13:
> Same issue here after updating to v1.10.0 this morning. In my case I think it's when I have more than 1 plate in my design file.
> 
> If I have all my models on a single plate and hit slice, everything is good. If I add a new plate and drag 1 model to it then slice that plate, performance drops to like 1fps and I can no longer pan around the camera on the preview tab.
> 
> I'm on macOS Sonoma 14.6.1.
> 
> I rolled back to the previous version.

thanks all for reporting this issue

@sagatxxx @northdakota @aayevtushenko @HO44-PROJECT @mikegabriel @zanuda2raza @WangNingkai 

do you also meet this issue with the same steps with @mikegabriel ?
if using only 1 plate, is the performance ok?

northdakota, 2024-11-13:
> thanks all for reporting this issue
>
> @sagatxxx @northdakota @aayevtushenko @HO44-PROJECT @mikegabriel @zanuda2raza @WangNingkai
>
> do you also meet this issue with the same steps with @mikegabriel ? [...]

Yes, after adding new plate - app become unresponsive

Haidiye00, 2024-11-13:
2,2 GHz 四核 Intel Core i5
16 GB 1600 MHz DDR3

lanewei120, 2024-11-13:
Hello @northdakota 
we have tried on our mac mini and macbook, with m1, m2, m3 chipshet
can not reproduce it

could you help to record a short video for the detailed steps
and could you help to upload the logs here?
(https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)

thanks very much

northdakota, 2024-11-13:
> Hello @northdakota we have tried on our mac mini and macbook, with m1, m2, m3 chipshet can not reproduce it
> 
> could you help to record a short video for the detailed steps and could you help to upload the logs here? (https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)
> 
> thanks very much

Sure, hope it will be helpful

https://www.youtube.com/watch?v=LSnDuKef6z4

[logs.zip](https://github.com/user-attachments/files/17725170/logs.zip)

Haidiye00, 2024-11-13:
Hello, can you switch the language to English, then close the software and reopen it to see if it works properly

CodeByZach, 2024-11-13:
> Hello, can you switch the language to English, then close the software and reopen it to see if it works properly

I have the exact same issue and I'm using english.

northdakota, 2024-11-13:
> Hello, can you switch the language to English, then close the software and reopen it to see if it works properly

I've swithed to English, restart app and issue still happened.

Also i've tried to create a new project instead of using old one.

- Created new project
- added random stl file to single plate
- slice
- all ok
- create a second plate
- copy and paste model from plate 1 to plate 2 
- slice
- issue reproduced again

northdakota, 2024-11-13:
Also i don't see any anomalies with CPU and GPU 
First spike - it's opening Bambu Studio
When reproduce the problem - no spikes

[image]

Haidiye00, 2024-11-13:
[image]
[image]
Hello, you can take a screenshot according to this. Thank you

Haidiye00, 2024-11-13:
[image]
[image]
Hello, can you help test this method? Follow the screenshot to close the two boxed options, then restart the software and help test if there are still any issues? thank you

HO44-PROJECT, 2024-11-13:
Hello, I just woke up (UTC+1) and discovered I'm not alone!
Thanks for suggestions and investigating.

I disabled these two boxes, restarted the software.
Open, slice... very quickly
Once in preview, the problem appears.
View change (i.e., preview to device) is about 5 seconds.
Unable to move the plate with the mouse in preview

FiredEgg-bambu, 2024-11-13:
@HO44-PROJECT can you please take a try to change the resolution or reduce the refresh rate？ Thanks for your help！

HO44-PROJECT, 2024-11-13:
Changed from 60 Hz to 50 Hz, problem is here
Changed to 720p, problem is here
Changed to another resolution, problem also.

I've restarted the studio between tests

The problem still appears once in preview, just after slicing.

HO44-PROJECT, 2024-11-13:
My complete configuration. I have two displays. I made the test with either one or two connected

[image]

Haidiye00, 2024-11-13:
> 嗨，我一直在跟踪涵盖此问题的两个线程上的一些对话，我没有取消选中版本 1.10.0.89 中记下的两个框，它仍然在预览屏幕中显示相同的行为 - 移动缓慢、软件挂起与 和 相同。
> 
> 这仅发生在具有多个构建板的项目中。
> 
> 我也在 Intel Core i9 2019 iMac 上，显示屏为 50 Hz。 ![build 1 10 0 89 hang](https://private-user-images.githubusercontent.com/41279/385647430-7e55c82a-8549-4092-a61f-a884414f84ad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0ODQ4OTcsIm5iZiI6MTczMTQ4NDU5NywicGF0aCI6Ii80MTI3OS8zODU2NDc0MzAtN2U1NWM4MmEtODU0OS00MDkyLWE2MWYtYTg4NDQxNGY4NGFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEzVDA3NTYzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4NzVlOGM5MzE0MGExYzY5ZWQ2YjA5Y2QwMTdkMzcwZjBkMWQyNTViN2M0OGIxNjQxYzUwZGRjNmQ0ZmNlZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.oiXiLW3fp5LG4D7H8SAgnpVYydQD6Xo_uYgdJD2SRTE)

Is it normal in the prepare interface for multi disk projects that are not sliced？
在准备页面的多盘项目是否正常呢？是只有预览界面的多盘项目有问题？

FiredEgg-bambu, 2024-11-13:
@HO44-PROJECT @nakedgremlin 
Could you try out our 1.10 beta version? We want to identify which version introduced this issue. Thank you very much!
1.10beta2: https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.81/Bambu_Studio_mac-v01.10.00.81-20241028192521.dmg
1.10beta1 : https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.74/Bambu_Studio_mac-v01.10.00.74-20241015212038.dmg

HO44-PROJECT, 2024-11-13:
I tried the 1.10beta2 link which is a 1.10.0.81, same issue with this advice:
[image]

I tried the 1.10beta1 link which is a 1.10.0.74, same issue

With both versions, problem still here

nakedgremlin, 2024-11-13:
I'm confirming as well both of these show the exact same issue when you have more than one plate in Preview:

- 1.10beta2 (v01.10.00.81): Sluggish and hangs
- 1.10beta1 (v01.10.00.74): Sluggish and hangs

When you delete the second plate, it performs just fine.

Haidiye00, 2024-11-13:
https://github.com/bambulab/BambuStudio/releases/tag/v01.09.07.52
Could you please test if this version is working properly? Thank you，I would like to confirm again, please

northdakota, 2024-11-13:
> https://github.com/bambulab/BambuStudio/releases/tag/v01.09.07.52 Could you please test if this version is working properly? Thank you，I would like to confirm again, please

[image]

On this version - all looks fine. No hung

FiredEgg-bambu, 2024-11-13:
@HO44-PROJECT @northdakota 

Can you provide your email address? We will send you some beta installation packages, and we need your help to locate this issue. Thank you very much

northdakota, 2024-11-13:
> @HO44-PROJECT @northdakota
> 
> Can you provide your email address? We will send you some beta installation packages, and we need your help to locate this issue. Thank you very much
Sure.

pol.vova@gmail.com

TransparentHouseSF, 2024-11-13:
> ![img_v3_02gj_8b065a80-fcbe-4f51-b833-dcee96824e0g](https://private-user-images.githubusercontent.com/132528822/385634883-33fab8e1-1f27-4a99-a330-0179dcd1d840.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0OTA1OTQsIm5iZiI6MTczMTQ5MDI5NCwicGF0aCI6Ii8xMzI1Mjg4MjIvMzg1NjM0ODgzLTMzZmFiOGUxLTFmMjctNGE5OS1hMzMwLTAxNzlkY2QxZDg0MC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExM1QwOTMxMzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMTg1N2ExZDRkYzZhMWJlYTY0NzRhYTNkMDcyZDY5MWFkYjEwNzEzNTY4NzMyMTljNWNmNDNlNjU4ZjBhMzM0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.81FPxjai9QeMZF8yIExmSHrSutmKigdEDf0Tp4tSpm8) ![img_v3_02gj_8a9692e8-7464-4671-a5bd-5751a1eb01ag](https://private-user-images.githubusercontent.com/132528822/385634913-76936360-a4e4-4ae3-9719-1219bd6b6d60.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0OTA1OTQsIm5iZiI6MTczMTQ5MDI5NCwicGF0aCI6Ii8xMzI1Mjg4MjIvMzg1NjM0OTEzLTc2OTM2MzYwLWE0ZTQtNGFlMy05NzE5LTEyMTliZDZiNmQ2MC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExM1QwOTMxMzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZTk0MTc1YTkxZTk3ZmQxYTkyOThhNDUzM2YyMDViZjg1ZTQ3MWU5NWZjYzE5NTM1NWE3NjQyMzA3N2M4OGQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.7Zygwt4BK1mEPfsZz6k3fyZ_GcRCKV-sImO_4JsvokA) Hello, can you help test this method? Follow the screenshot to close the two boxed options, then restart the software and help test if there are still any issues? thank you

dosnt help at all. same issue. BS hangs after click Slice with project on multiple build plates

MacOS 14.7.1
intel iCore i9 
AMD Radeon Pro 5500M 8 GB
Intel UHD Graphics 630 1536 MB

HO44-PROJECT, 2024-11-13:
> @HO44-PROJECT @northdakota
> 
> Can you provide your email address? We will send you some beta installation packages, and we need your help to locate this issue. Thank you very much

ho44project@gmail.com

HO44-PROJECT, 2024-11-13:
got your mail, test in progress, thanks

HO44-PROJECT, 2024-11-13:
1.10.0.50 same issue

northdakota, 2024-11-13:
1.10.0.50 issue present

HO44-PROJECT, 2024-11-13:
1.0.10.68 same issue too, sorry

HO44-PROJECT, 2024-11-13:
10.0.0.68 same issue too, sorry.

HO44-PROJECT, 2024-11-13:
I noticed that if I go back to "prepare" tab then "preview" tab then I can now move the object

This is my test case:

1. 1.10.0.68 fresh install
2. Open an existing project with multiple plates
3. Slice the first plate, studio goes to preview
4. Application doesn't answer, can't move the object in space
5. Takes about 30s to go back to "prepare" view while showing the "spinning wheel cursor"
6. Choose the view tab
7. Application is now answering

qzbeta, 2024-11-13:
The same situation occurs on Macs with Intel CPUs.
When slicing is completed, staying on the preview interface will become unresponsive.
If you can click Return to the Preparation Page, it will return to normal.
Temporary solution: Click Return to the Preparation Page as soon as possible before slicing is completed.

mikegabriel, 2024-11-13:
I reinstalled 1.10.0.89 this morning and did a bit more testing and for me it does seem to take a significant performance drop the second I add an additional plate.  

If I start with an empty file, right click the build plate and add a primitive -> 3D benchy and then slice, everything performs normally.

Now If I click Add Plate and then reclick my first plate so it's active, drag the benchy so that I can click Slice plate again, it's completely unusable in the preview tab and beachballs.

Here's a video of the above: https://github.com/user-attachments/assets/9c523c28-ad98-4f28-904b-ab95ac7c98ac

[image]

refk1, 2024-11-13:
I had a huge lag problem (spinning pizza) and similar issues as others mentioned. I restarted my Mac and still had the same issue. I restarted again and with no apps open, I ran the CleanMyMac app and opened the slicer and now everything seems to be working like normal. Perhaps it's cache issue. I was also seeing issues like the attached image. Have not tested to see if it's still doing it after running that app. 
[image]

fahaaaxiki, 2024-11-14:
**V1.10.0 Hotfix Preview**
**Windows**:https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
**Mac**:https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg
We have urgently fixed issues including STEP import crashes, lag on Mac(Intel CPU) after slicing, and temperature display errors. Please use the above version to resolve these issues you may encounter before we officially release the Hotfix version.

HO44-PROJECT, 2024-11-14:
I was lucky enough to be able to help the Bambu guys who worked late to figure out the case. They are friendly and worked late and hard and found the cause. Stay tuned.

mikegabriel, 2024-11-14:
Just confirming that the above hotfix preview appears to have resolved the issue on my end.  I've opened a few files that all were giving issues yesterday and they are working today.

CodeByZach, 2024-11-14:
Talk about a timely response and resolution! More often than not, companies like Apple and Sony allow customer feedback to go unheard. So as not to take for granted, I think it's worth commending when we get things right!

Cheers :)

Shoutout to who I'm assuming are the Bambu Devs @Haidiye00 @lanewei120 @FiredEgg-bambu @fahaaaxiki 
And of course @HO44-PROJECT @mikegabriel @WangNingkai @zanuda2raza @aayevtushenko @sagatxxx (pretty sure I've seen your stuff on MakerWorld, cool stuff) @northdakota @nakedgremlin @TransparentHouseSF @QZBT @refk1 @kettleblaze

QZBT, 2024-11-14:
> **V1.10.0 Hotfix Preview V1.10.0 修补程序预览** **Windows**:https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe**Windows** ： [https](https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe) ://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe **Mac**:https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmgMac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg We have urgently fixed issues including STEP import crashes, lag on Mac(Intel CPU) after slicing, and temperature display errors. Please use the above version to resolve these issues you may encounter before we officially release the Hotfix version.我们紧急修复了STEP导入崩溃、切片后Mac（Intel CPU）上的滞后以及温度显示错误等问题。在我们正式发布Hotfix版本之前，请使用上述版本解决您可能遇到的这些问题。

This version works fine. Thank you very much for the prompt fix.

HO44-PROJECT, 2024-11-14:
I do agree with all of you. Thanks a lot !

aayevtushenko, 2024-11-14:
> Hello, we suspect it may be a problem with the graphics card or graphics card driver. Could you please provide your graphics card model and driver? Thank you

AMD Radeon Pro 5300M 4 GB
Intel UHD Graphics 630 1536 MB

on MacBookPro 2019.

The hotfix seems to have resolved the responsiveness problem.

aayevtushenko, 2024-11-14:
update: updating to the new versions seems to have removed all filament profiles and slicing profiles I have either created or loaded. 🐞

closed 2024-11-15 by HO44-PROJECT

sxdjt, 2024-11-20:
Confirming that 1.10.1.50 fixed the issue... and is *much* faster when slicing multiple plates.

Thank you Bambu!