bambulab/BambuStudio #5195: Filament Change Applies to Plug Connector Instead of Added Text in Cut Objects

bbbenji, 2024-11-12:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Built from source by yourself or others

### OS version

Linux nixos 6.6.60 #1-NixOS SMP PREEMPT_DYNAMIC Fri Nov  8 15:28:28 UTC 2024 x86_64 GNU/Linux

### Printer

Bambu Lab X1 series

### How to reproduce

1. Open Bambu Studio and load a model containing a cut object with plug connectors.
2. Add text to the cut object that has the plugs, using the text tool.
3. Attempt to change the filament type/color specifically for the added text.

### Actual results

The filament change is applied to one of the plug connectors instead of the text.

### Expected results

The filament change should be applied to the text instead of one of the plug connectors.

### Project file & Debug log uploads

[bug.zip](https://github.com/user-attachments/files/17718652/bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-11-13:
[image]
We have identified the issue and resolved it. Thank you for your discovery. Thank you

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00