bambulab/BambuStudio #5196: Temperature informations and finish time do not work remotely

JuanitoTaco, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Printer

Bambu Lab X1

### How to reproduce

Open Bambu Studio in remote, go to your printer

### Actual results

[image]

### Expected results

Have those informations please

### Project file & Debug log uploads

[Remote temperature problem.zip](https://github.com/user-attachments/files/17719162/Remote.temperature.problem.zip)
[log.zip](https://github.com/user-attachments/files/17719166/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

FallasAutomation, 2024-11-12:
Same problem.

Further testing shows that when the current temp changes, the temperature is displayed.

[image]

[image]

FallasAutomation, 2024-11-14:
Hot fix released:

https://github.com/bambulab/BambuStudio/issues/5264

DanBao-Bambu, 2024-11-19:
The reason for the problem with the completion time may be due to the screen zoom ratio. Please change the screen zoom ratio or try using a new version of Studio to solve this problem. If you have any questions, please provide feedback here.

Tumsk, 2024-11-19:
> The reason for the problem with the completion time may be due to the screen zoom ratio. Please change the screen zoom ratio or try using a new version of Studio to solve this problem. If you have any questions, please provide feedback here.

Yes, this bug is linked to Windows Display Settings > Scale.
100% (Recommended) - Finish Time **not** displayed.
125% - Finish Time displayed
150% - Finish Time displayed

So Finish Time not displayed bug only happens at 100% (Recommended) display scale.

Tested on Bambu Studio v1.10.1.50