bambulab/BambuStudio #5197: STEP File Import Fail

smartLars, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Gerätename	AMD
Prozessor	AMD Ryzen 9 5900X 12-Core Processor               3.70 GHz
Installierter RAM	64,0 GB
Systemtyp	64-Bit-Betriebssystem, x64-basierter Prozessor

### Printer

X1, A1, A1 Mini

### How to reproduce

1. neues Projekt anlegen
2. Hinzufügen
3. STEP-Datei auswählen
[image]

Egal was dann ausgewählt wird, es passiert nichts oder das Programm stürzt ab.

### Actual results

Egal was dann ausgewählt wird, es passiert nichts oder das Programm stürzt ab.

### Expected results

Datei sollte geladen werden können.

### Project file & Debug log uploads

ich kann kein finales Projekt hochladen, da ja keine Daten importiert werden

### Checklist of files to include

- [ ] Log file
- [ ] Project file

benjaminkott, 2024-11-12:
please see https://github.com/bambulab/BambuStudio/issues/5186

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu