bambulab/BambuStudio #5198: Broke STEP Import with V10

MonsterMaxx, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

i912950, 64GB of Ram, Quadro A4500

### Printer

X1C

### How to reproduce

Just upgraded to V10 and STEP import is no longer succeeding.

First on default it was radically reducing resolution which I definitely do NOT want.

So I unchecked that option in settings and reimported models. It’s just sitting there spinning on my fastest machine. CPU is at 100% and it’s not making any progress after 10 minutes.  I finally killed the task after 20 min.

So I went over to my daily driver which is slower, but has the latest of 1.9.7.52 and everything worked normally.

Looks like BL has broken the STEP file import process and I’ll have to stick with V1.9 for now.

### Actual results

unresponsive

### Expected results

import models

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Stevelsch, 2024-11-14:
https://github.com/bambulab/BambuStudio/issues/5261

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

MonsterMaxx, 2024-12-14:
For me, V10 step file import is totally broken now.  I have to use V9 to import