bambulab/BambuStudio #5199: V 1.10 "Unhandeld unknown exception:"

bigcpb, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel i7

### Printer

Bambu Lab A1 mini

### How to reproduce

1.  try to open Project stored in *.step formate
2.  Step file import parameters windows (tile) opens whilst loading the project-file "*.step"
3. changing one of the parameter the tile reports will lead to ":Calculating, please wait" but it will not finish
4.

### Actual results

when i change the second parameter or press the OK button, the main window closes, only a tile with an error message remains on the screen
[image]
[image]

### Expected results

modify import parameter

### Project file & Debug log uploads

[debug_Tue_Nov_12_17_53_46_27972.log.zip](https://github.com/user-attachments/files/17720380/debug_Tue_Nov_12_17_53_46_27972.log.zip)
[Kabelhal
[Clip gelber Sack.zip](https://github.com/user-attachments/files/17720413/Clip.gelber.Sack.zip)
ter.zip](https://github.com/user-attachments/files/17720404/Kabelhalter.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu