bambulab/BambuStudio #5200: New version terribly slow and unresponsive after slicing

aayevtushenko, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.7

### Additional system information

Processor: 2.6GHz 6-Core Intel Core i7
Memory: 16GB
macOS: Sonoma 14.7

### Printer

Bambu A1

### How to reproduce

Slice this model https://makerworld.com/en/models/746690#profileId-679824

Try moving camera, clicking menus, etc

### Actual results

Any action takes 7+ seconds to execute

### Expected results

like with the previous version actions should work without painful lag

### Project file & Debug log uploads

log file is not available at application support/bambustudio.
[Retractable_Cable_Organizer_V2.3mf.zip](https://github.com/user-attachments/files/17720618/Retractable_Cable_Organizer_V2.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zanuda2raza, 2024-11-12:
Same MBP 16, i9, macOS 15.1

FEMit160, 2024-11-12:
Seems to work the same (good) as prior versions for me. Windows 10, i7-9th gen, 16GB ram, GF GTX 1650 4gb mobile.

mikegabriel, 2024-11-12:
Based on comments I have read I think it's an Intel / Mac thing.  Over on the reddit subreddit someone replied to my comment that it's fine on their M1 Pro.

For the record, I am on a 2019 MBP with an Intel I9 and 32gb ram, 8gb ram radon 5500m.  

Rolling back to previous version resolved the issue.

Cyberdoc, 2024-11-12:
Same here: 2019 MBP, i9, 64GB RAM, 8GB 5500m - macOS 15.0.1

Also back to previous version, which "removed" the problem.

okovpashko, 2024-11-12:
I just updated to v1.10 and noticed the same issue.
2023 MacBook Pro M3 Max, 36 GB RAM, macOS Sonoma 14.16.1.

Also, it seems like the organic tree support generation is broken as it generates the default "normal" support.

Had to roll back to v1.9.7.52 where everything works fine.

fah, 2024-11-12:
iMac Intel/2019: Same problem. I only see the ball spinning every time I move the cursor. Not usable!
V 1.10.0.89

killsecurly, 2024-11-12:
This all seems to be a Intel Mac issue. My MBA M1 works fine with v1.10.

CodeByZach, 2024-11-13:
Same exact issue here. Slicing makes it unusable. Had to revert.

lanewei120, 2024-11-13:
the same as https://github.com/bambulab/BambuStudio/issues/5194

Haidiye00, 2024-11-13:
Hello, we suspect it may be a problem with the graphics card or graphics card driver. Could you please provide your graphics card model and driver? Thank you

lanewei120, 2024-11-13:
Hello, @mikegabriel @Cyberdoc @CodeByZach @okovpashko @aayevtushenko @zanuda2raza @FEMit160 @killsecurly 
thanks for reporting this issue

we have tried our best to reproduce this issue here,
but still can not reproduce it

could someone also help to record a short video for the detailed steps
and help to upload the log here(https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)
just like [northdakota](https://github.com/northdakota) did in issue 5194

we need to collect more details to dig into this issue

Haidiye00, 2024-11-13:
[image]
[image]

Hello, can you take a screenshot according to this? Thank you

tangketan, 2024-11-13:
@okovpashko As to the support issue, did you use variable layer height? Organic support doesn't work with variable layer height, so we defaulted to hybrid tree support, which may generate "normal-like" structures.

Haidiye00, 2024-11-13:
[image]
[image]
Hello, can you help test this method? Follow the screenshot to close the two boxed options, then restart the software and help test if there are still any issues? thank you

Haidiye00, 2024-11-13:
https://github.com/bambulab/BambuStudio/releases/tag/v01.09.07.52
Could you please test if this version is working properly? Thank you

TransparentHouseSF, 2024-11-13:
> https://github.com/bambulab/BambuStudio/releases/tag/v01.09.07.52 Could you please test if this version is working properly? Thank you

this old version is working fine. 1.10.0.89 is not.

okovpashko, 2024-11-13:
@tangketan I'm sorry, I mixed up the organic tree support with the hybrid. The organic tree works as expected. The hybrid tree in v1.10 is generated as "normal" whereas v1.9.7 generates it as the "tree". When I open the same project with these two versions of Bambu Studio I see different support on the preview tab.

I didn't find any related changes in the changelog so I'm wondering if it's an expected behavior.
Should I open a separate issue for that?

okovpashko, 2024-11-13:
As for the main issue of the current topic, I'm confused now.
Yesterday I definitely had a slow and irresponsive rendering on the preview tab in v1.10 and replaced it with 1.9.7 where everything worked well. Today I installed v1.10 in parallel to v1.9.7 and there are no issues with the preview at all 🤔 

The v1.10 was installed on a fresh OS that never had the previous Bambu Studio versions.

mkim1, 2024-11-13:
I'm also having this issue with the following device. Downgrading to v01.09.07.52 resolved the issue for me.
[image]

mikegabriel, 2024-11-13:
I reinstalled 1.10.0.89 this morning and did a bit more testing and for me it does seem to take a significant performance drop the second I add an additional plate.  

If I start with an empty file, right click the build plate and add a primitive -> 3D benchy and then slice, everything performs normally.

Now If I click Add Plate and then reclick my first plate so it's active, drag the benchy so that I can click Slice plate again, it's completely unusable in the preview tab and beachballs.

[image]

Malrikgus, 2024-11-13:
Same issue here. Monterey 12.7.

QZBT, 2024-11-14:
The same situation occurs on Mac with Intel CPU.
When slicing is completed, staying on the preview interface will become unresponsive.
If you can click Return to the Preparation Page, it will return to normal.
Temporary solution: Click Return to the Preparation Page as soon as possible before slicing is completed.

Batou99, 2024-11-14:
Same here, iMac with intel i5, Radeon Pro 580X 8 GB. Completely unusable.

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

aayevtushenko, 2024-11-14:
the hotfix provided in this other issue seems to have resolved the issue.
https://github.com/bambulab/BambuStudio/issues/5194#issuecomment-2476700409

aayevtushenko, 2024-11-14:
Can this issue be merged into [a duplicate](https://github.com/bambulab/BambuStudio/issues/5194#issue-2652498309)?

HO44-PROJECT, 2024-11-15:
They just released 1.10.1.50 that closes my similar case.
https://github.com/bambulab/BambuStudio/issues/5194

closed 2024-11-18 by Haidiye00