bambulab/BambuStudio #5201: V1.10 Splitting STL into multiple objects does not retain colors properly

FEMit160, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Window 10

### Additional system information

CPU: Intel Core i7 9th gen, Mem: 16GB, Graphics: Geforce GTX 1650 4GB (mobile)

### Printer

Bambu Lab A1

### How to reproduce

1. Open a precolored filed in Bambu Studio (this is one example that I tested https://makerworld.com/en/models/206695#profileId-226815)
2. Select Model that you would like to split into multiple objects or parts.
3. Select either "Split to Objects" or "Split to Parts"

### Actual results

Some color is retained and others are lost. Also, wrong colors are assigned to some triangles
[image]

### Expected results

All colors retained properly after splitting
[image]

### Project file & Debug log uploads

[BrokenColorsAfterSPlit.zip](https://github.com/user-attachments/files/17721090/BrokenColorsAfterSPlit.zip)
[debug_Tue_Nov_12_09_33_10_5616.log.zip](https://github.com/user-attachments/files/17721092/debug_Tue_Nov_12_09_33_10_5616.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-11-13:
Can you provide the original file? Thank you

FEMit160, 2024-11-13:
[NCC-17001.zip](https://github.com/user-attachments/files/17725000/NCC-17001.zip)
Original File

Haidiye00, 2024-11-13:
[image]
Thank you for discovering the issue. I have fixed it. Thank you~

FEMit160, 2024-11-13:
> ![image](https://private-user-images.githubusercontent.com/132528822/385630437-9eb0dc2b-3903-4e3d-9f3e-29d9d08ca85f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Wow, that was fast! Thank you for continueing to support pre-colored model features! It definitely sets Bambu apart from other slicers!

Haidiye00, 2024-11-13:
At present, you can import STL first, split it first, and then paint it. Bug fixes need to wait for the new version. I will notify you once the version is released. Thank you

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00