bambulab/BambuStudio #5202: V1.10 Import Step File "Import Parameter" Popup - Can't See Quality Circles in Dark Mode

FEMit160, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Window 10

### Additional system information

CPU: i7-9th gen, Mem: 16GB, Graphics, Geforce GTX 1650 4GB Mobile

### Printer

Bambu Lab A1

### How to reproduce

1. Go to Preferences>Enable Dark Mode
2. File>Import>Any Step File of your choosing

### Actual results

Can't see quality circles since they are black lines

[image]

### Expected results

Should be able to see white lines for the circles when in dark mode, like you can see black lines in non dark mode.
[image]

### Project file & Debug log uploads

Any step file will do. Sorry, I don't have a non propriety step to upload at the moment

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-11-13:
confirm

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50