bambulab/BambuStudio #5203: V1.10 Import Step File, program crashed / Apple M2 Pro

Vespisti1969, 2024-11-12:
### Bambu Studio Version

1.10.0

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS 15.0.1 (24A348)

### Additional system information

Apple M2 Pro

### Printer

Bambulab X1C Combo with 2 AMS

### How to reproduce

1. Open Bambu Studio
2. creat new projekt and load the .step-file
3. Mesh-Window open and after few seconds the program crashed and closed

### Actual results

Download and install the older Version: 1.9.7.52 Public Official

### Expected results

-

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Crushnik88, 2024-11-12:
Same for me, upon importing step file this pops up and crashes the app as soon as i move the second swipe bar.
[image]

benjaminkott, 2024-11-12:
please see https://github.com/bambulab/BambuStudio/issues/5186

Vespisti1969, 2024-11-12:
But that cannot be the solution. There are people who do not speak or understand English.

benjaminkott, 2024-11-12:
please see the replies https://github.com/bambulab/BambuStudio/issues/5186#issuecomment-2470555626

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

Vespisti1969, 2024-11-14:
Yes, and it works Great. No issues when you load a step , if german Language is set.

closed 2024-11-18 by MackBambu