bambulab/BambuStudio #5207: Seam Position seems to be ignored

Naiki92, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.0.1

### Printer

Bambu Lab X1C/ Bambu Lab A1 Mini

### How to reproduce

1. Import Model
2. Define `Seam Position: Aligned`
3. Slice
4. Look how the seams are aligned
5. Now switch to `Seam Position: Back`
6. Slice
7. Look if the seams have changed

### Actual results

For me the seams are changed but not at the back as expected.

### Expected results

Seams should be at the back

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2024-11-13:
[image]
I have tried it with my Mac(2022,M2), and it seems all right, can you offer a 3mf for me to test?

Naiki92, 2024-11-13:
[seam_test.3mf.zip](https://github.com/user-attachments/files/17730895/seam_test.3mf.zip)

MartinChan3, 2024-11-13:
I have received your 3mf and test it, the result is due to Scarf Seam was open by default (This is a new feature of V1.10.0), and the Scarf Seam rendering is different from that of original seam. If you want the old seam you can turn off the scarf seam like following:
[image]

Naiki92, 2024-11-13:
Ah, great, thanks 👍 That helped