bambulab/BambuStudio #5209: Reload from disk does not work anymore for step files

benjaminkott, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Load a step file
2. Change original step file
3. Reload from disk

### Actual results

Model is not updated

### Expected results

Model is updated as before

### Project file & Debug log uploads

.

### Checklist of files to include

- [X] Log file
- [X] Project file

MrDB42, 2024-11-12:
Confirm bug.

MackBambu, 2024-11-13:
confirm

rashuge, 2024-11-13:
Confirm

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

jonatkins, 2024-11-14:
I had the same issue. That new build (1.10.0.94) has fixed it for me.

closed 2024-11-18 by MackBambu