bambulab/BambuStudio #5211: camera cam initialization problem

tsem006, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

13th Gen Intel(R) Core(TM) i7-13700   2.10 GHz - Nvidia 4800TI - 32.0 GB (31.8 GB utilizzabile) - Nvidia 4070 TI

### Printer

Bambu Lab P1S

### How to reproduce

bambu studio does not display the webcam, the initialization message remains and it does not show the camera

### Actual results

bambu studio does not display the webcam, the initialization message remains and it does not show the camera

### Expected results

[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-11-13:
Try restarting the printer and replaying.

tsem006, 2024-11-13:
Ho provato diverse volte, ho reinstallato Bambu studio, ho aggiornato
windows ma nessun cambiamento.
Può derivare dal fatto che non ho aggiornato il firmware della stampante
P1S? ho sentito diversi problemi derivati dall'aggiornamento alla 1.6 e
quindi non l'ho mai aggiornata (1.5).

Saluti
luca

bambu123, 2024-11-13:
If the problem persists, please provide log files.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
> ***@***.***>
> Data:  mer, 13 nov 2024 15:31
> Oggetto:  Re: [bambulab/BambuStudio] camera cam initialization problem (Issue #5211)
> ***@***.***>
> ***@***.***>, ***@***.***>
> 
Ho provato diverse volte, ho reinstallato Bambu studio, ho aggiornato

> 
windows ma nessun cambiamento.

> 
Può derivare dal fatto che non ho aggiornato il firmware della stampante

> 
P1S? ho sentito diversi problemi derivati dall'aggiornamento alla 1.6 e

> 
quindi non l'ho mai aggiornata (1.5).

> 
> 
Saluti

> 
luca

> 
> 
> 

ha scritto:

MartinChan3, 2024-11-13:
@tsem006 I have tried it with my P1S and not only in printing or not the liveview works well.
[image]
[image]