bambulab/BambuStudio #5214: Issue with Coloring by Component in Bambu Studio 1.10 Update

Sitsero, 2024-11-12:
### Bambu Studio Version

1.10

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

X1/A1

### How to reproduce

Import a .STEP file with multiple bodies per component and try coloring by component. 

### Actual results

After updating to Bambu Studio 1.10, I noticed that the coloring options have changed. Previously, I could color my model by component, which was really helpful. Now, it only shows options to color individual bodies within each component, which is a bit frustrating.

For example, in the screenshot attached, you can see that components with only one body (Green in this instance) still show up like they did before the update, allowing me to color them as a whole. But any component with multiple bodies is divided into individual bodies for coloring, which is similar to what happens when importing 3MF files.

Is there a way to get back to component-level coloring? Or any workaround that lets me apply color to entire components at once instead of each body individually? Any help or suggestions would be greatly appreciated!

Thanks!
[image]

### Expected results

The objects shoulkd be seperater by component and not by body, like before. Or at least a toggle option to chose from?

### Project file & Debug log uploads

[Oshawott Lightbox.zip](https://github.com/user-attachments/files/17723569/Oshawott.Lightbox.zip)
Please keep the file confidential as it is an unreleased project. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-11-13:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you

Sitsero, 2024-11-13:
> Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you

Sure, here are the segmented components.
[image]

Sitsero, 2024-11-13:
FYI, the components on my end are as shown in the picture below, I've expanded one of them just to show that for example Black-SOLID-1 that shows up in the GIF is actually the Body1 inside the Black component. 
[image]
How it should show up is demonstrated below, when making the same project with OrcaSlicer
[image]

MackBambu, 2024-11-13:
It seems that an option is needed to confirm whether the assembly of the model should be further decomposed, as some users have this requirement.
https://github.com/bambulab/BambuStudio/issues/4560

fahaaaxiki, 2024-12-03:
Hello, we are working on this problem. We want to use the original .step file to verify whether the fix is effective. Could you please provide the original file？Thanks.

Sitsero, 2024-12-03:
> Hello, we are working on this problem. We want to use the original .step file to verify whether the fix is effective. Could you please provide the original file？Thanks.

Yes of course, here it is. Since I couldn't upload a .STEP file here, I've uploaded it in the published model's additional files section. Here's the link: https://makerworld.com/en/models/783738#profileId-721189

pdls3d, 2024-12-16:
Any updates? This is a pretty significant issue for complex geometry that has been strategically designed for step file object layers.

Sitsero, 2024-12-16:
> Any updates? This is a pretty significant issue for complex geometry that has been strategically designed for step file object layers.

No idea, I'm just using Orca as long as the issue persists, saving the project and then opening it with Bambu and resaving it so it can be uploaded on Makerworld.

cowboy, 2024-12-26:
Thanks for filing this issue, I'm experiencing the same problem with 1.10.1, and reverting to 1.9.7.52 fixed it.

MackBambu, 2025-01-11:
fix issue
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.64

closed 2025-01-12 by MackBambu