bambulab/BambuStudio #5215: Upload missing on Bambu studio 1.10.0.89

tsem006, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU i7 - 32 mb windows 11 - nvidia 4050 ti

### Printer

Bambu Lab P1S

### How to reproduce

On Preview mode

### Actual results

I noticed that there is no upload button in the new version, is it a bug or wanted?

### Expected results

ripristinare il pulsante

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

andyngo, 2024-11-13:
They've moved it into a sub-menu:
[image]

tsem006, 2024-11-13:
> They've moved it into a sub-menu:
> 20241113_082010AM_Screen_Capture_Safari.png (view on web)

Perfetto!, grazie della tempestiva risposta.

Saluti
luca

LiZ-Li-BBL, 2024-11-19:
This is indeed a intentional change. The issue could be closed if you have confirmed :)

tsem006, 2024-11-19:
Certo puoi chiudere  la questione
Saluti
Luca