bambulab/BambuStudio #5216: Temp Readouts not showing correctly on Device page after latest update

HomerSimpy, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1

### Additional system information

M3 MAX MacBook Pro, 48 GB

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Open Bambu, go to device page, Temps are incorrect or not reading at all, start print, still not right

### Actual results

[image]

### Expected results

correct temp readings from hot end, bed  and chamber

### Project file & Debug log uploads

[Temperature Issue.3mf.zip](https://github.com/user-attachments/files/17723883/Temperature.Issue.3mf.zip)
[debug_Tue_Nov_12_17_33_55_1093.log.0 copy.zip](https://github.com/user-attachments/files/17723904/debug_Tue_Nov_12_17_33_55_1093.log.0.copy.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-11-13:
hello @HomerSimpy 
thanks for reporting this issue

is it working before when using previous bambu studio version?
we are wording whether it is a new issue caused by 1.10.0.89 or not

HomerSimpy, 2024-11-13:
It was working as expected prior to the update.  The first number which I
will call the “Set Point”, or desired temp was always visible. The number
after the forward slash (/) was always the “Actual” temp which by nature
should be dynamic. I haven’t downgraded yet to double check but I’m 99%
sure.

Paul B. Breeden Sr.
(603) 560-163
***@***.***

FallasAutomation, 2024-11-13:
Bug reported here:

https://github.com/bambulab/BambuStudio/issues/5196