bambulab/BambuStudio #5218: SE CIERRA BAMBU STUDIO

ALBERTOBARRAGAN1983, 2024-11-12:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

MACOS MONTEREY 12.7.6

### Printer

P1S

### How to reproduce

1- ABRO EL PROGRAMA BAMBU STUDIO
2- CREO NUEVO PROYECTO
3- AÑADO ARCHIVO FORMATO STEP
4- SE ABRE UNA VENTANA DE step file import parametro
[image]
s
5- SE CIERRA AUTOMÁTICAMENTE EL PROGRAMA DANDO ERROR
6- REPITO EL PROCESO INDICADO Y SIGUE HACIENDO LO MISMO

### Actual results

[Piezas cables ordenador.zip](https://github.com/user-attachments/files/17724001/Piezas.cables.ordenador.zip)

### Expected results

Lo que se espera es que no se cierre el programa 

### Project file & Debug log uploads

A
[Piezas cables ordenador.zip](https://github.com/user-attachments/files/17724000/Piezas.cables.ordenador.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-11-13:
please see https://github.com/bambulab/BambuStudio/issues/5186

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu