bambulab/BambuStudio #5220: Possible filament sync issue with AMS

dterracino, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Click synchronize filament list from AMS

### Actual results

Filaments do update based on filaments selected in AMS. However, this version shows the v arrow to the left of all filaments now, which in the past seemed to indicate some kind of sync issue.

[image]

### Expected results

Filaments should sync correctly and v arrow should disappear.

I'm including a project file, but it happens with any project file. Before the latest version, this would work correctly.

### Project file & Debug log uploads

[Triple Threaded Jar v1.zip](https://github.com/user-attachments/files/17725857/Triple.Threaded.Jar.v1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MartinChan3, 2024-11-13:
I'm not sure  'v arrow '  does something to affect the ams sync result or print result, It seem it should disappear after sync?

Zaphodjoe, 2024-11-18:
Since the update I can not Synchronize my AMS when I open Bambu Studio. I get the error
"No AMS filaments. Please select a printer in 'Device' page to load AMS info."
So once I go to Device then back I can reload the AMS filaments. Every time I open Bambu studio this happens. The old bambu studio used to have the printer ID on the prepare page so it saw the printer and was able to reload but now you have to do a whole extra step. Please put it back the old way. Other wise I guess its time to go to Orca.

DanBao-Bambu, 2024-12-02:
Please check if the option for multi device management is enabled in the settings. If so, synchronization may encounter issues.

Zaphodjoe, 2024-12-02:
[image]

Nope not enabled. Ever since the update(s) I did both that came out, it has been like this. Once I close Bambu Studio and re enter and am in Prepare, it can't see my AMS or SYNC until I go to Device and it refreshes my printer, then go back and it sees it. The prepare page no longer refreshes the printer for some reason.