bambulab/BambuStudio #5221: Seam painting offset

Strarrow, 2024-11-13:
### Bambu Studio Version

V 1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home

### Additional system information

Processor	AMD Ryzen 7 1700 Eight-Core Processor             3.00 GHz
Installed RAM	16.0 GB
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Click on model
2. Click on seam painting icon at the top
3. Paint vertical seam
4. Slice model
5. See seam is offset to right of where seam was painted

### Actual results

[image]
[image]
[image]
[image]

### Expected results

Seam should be where seam was painted.

### Project file & Debug log uploads

[Seam Painting Error Project File.zip](https://github.com/user-attachments/files/17726659/Seam.Painting.Error.Project.File.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MartinChan3, 2024-11-13:
It is due to 1.10.0 start to use Scarf Seam by default, the Scarf Seam rendering was not like common seam.You can close seam option under filament setting.
[image]

Strarrow, 2024-11-13:
Thank you, perfect solution!