bambulab/BambuStudio #5222: Ошибка в программе, сбой при попытке добавить файл STEP/STL.

Dimitr-Sibiryak, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 pro

### Additional system information

AMD Ryzen 9 7940HS w/ Radeon 780M Graphics 4.00 GHz, 

### Printer

Bambulab x1C

### How to reproduce

1. Add - 2. Ctrl+I - 3 (1). Chande parameters "Linear deflection" or "Angle deflection" - 4. press "Ok" - Unknown Exception/escape
3 (2). if no chaging - nothin happen wen press "Ok".

### Actual results

Error when trying to add a STEP/STL file: a new menu "STEP Import Options" appears. If you don’t change the options, the menu doesn’t respond to the "OK" button. If you change the options, an "Unknown Exception" error occurs, and the program closes.

### Expected results

SHOULD THE FILE BE UPLOADED FOR PRINTING, IS IT NOT CLEAR WITHOUT A DESCRIPTION???

### Project file & Debug log uploads

I can’t load the project file because Bambu Studio crashes when trying to load it. The project simply doesn’t exist; try it yourself. I’ll send the file in an archive, but there are no log files or project file records.
[Border AMG7200.zip](https://github.com/user-attachments/files/17726909/Border.AMG7200.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Dimitr-Sibiryak, 2024-11-13:
Why did I update the program…
Two old sayings:
The best is the enemy of the good. Don’t make things “better” for me; just leave them “good.”

MartinChan3, 2024-11-13:
The question is mentioned in #5186 , you can switch language to English to solve this problem temporarily.

zanuda2raza, 2024-11-13:
Смените язык на En и все заработает.

Dimitr-Sibiryak, 2024-11-13:
А я не хочу менять язык на английский.

zanuda2raza, 2024-11-13:
Тогда придется откатиться на 1.9.7 и ждать когда 1.10 починят. На macOS она вообще нерабочая

Dimitr-Sibiryak, 2024-11-13:
> Отправлено: Wednesday, November 13, 2024 10:00:07 AM
> Кому: bambulab/BambuStudio ***@***.***>
> Копия: Dimitr-Sibiryak ***@***.***>; Author ***@***.***>

Прикольно.
Так я в чате смотрю, там и немцы, и итальянцы, и французы с этим же сталкиваются, им тоже на английский переключаться? - особенно итальянцы с французами любят английский😂😂😂🤣🤣🤣

Дмитрий

zanuda2raza, 2024-11-13:
Да, и китайцам тоже. Но в этот раз они как-то накосячили сильно.

Dimitr-Sibiryak, 2024-11-13:
А вот и таблетка:
Я переключился обратно в русский, поднял линейную интерполяцию до значения 0,4, угловую до 0,2 и вуаля – он прожевал в русскоязычном интерфейсе файл STEP с русскими буквами в имени, и больше ошибок пока не возникало, несколько раз выгружал и загружал студио и без ошибок пока.

Походу во время компиляции у них кто-то на клаве уснул и соплями в значениях движков этих параметров ввёл недопустимый диапазон.

Кто бы им подсказал, где покопать…

Dimitr-Sibiryak, 2024-11-13:
> Отправлено: Wednesday, November 13, 2024 10:07:54 AM

Могу видео прислать, перешлите китайцам, пусть поржут.

Серебряков Д.В.

zanuda2raza, 2024-11-13:
у меня на компьютере с 2-мя Xeon Gold и 256 оперативки все разу идеально крутится, но оно жрет 45 ГБ оперы в этот момент. А на маке все колом встает. Они все видео просят на https://wiki.bambulab.com/en/x1/troubleshooting/upload-log и делать тикет

zanuda2raza, 2024-11-13:
Причем, что забавно, оно все равно работает лучше, чем Picaso-вские поделки....

Dimitr-Sibiryak, 2024-11-13:
> Отправлено: Wednesday, November 13, 2024 10:31:00 AM

Да, работает мьудио хорошо, но у меня ещё и креалити к1макс, и тут еитайцы отоичились и в наборе филаментов при выборе креалити указали только рпа, петг, абс и ещё пару каких-то, и не добавить.
Приходится по сути такой же оркаслайсер грузить для работы с креалити.

zanuda2raza, 2024-11-13:
Ну тут хотя бы выбор есть, а то с системой профилей в полигоне часов 6 разбирался.... а потом еще и выяснилось, что по умолчанию они сохраняются в формате для S1 принтеров и S2 их не читает...

Dimitr-Sibiryak, 2024-11-13:
> Отправлено: Wednesday, November 13, 2024 10:51:22 AM

Да какой там выбор, даже сраного силк пла нет, не говоря о всяких поликарбонатах, которыми иногда приходится баловаться...

Dimitr-Sibiryak, 2024-11-14:
Thanks!

Серебряков Д.В.

Отправлено: Thursday, November 14, 2024 6:31:47 PM

It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

zanuda2raza, 2024-11-14:
Видео, проверю) спасибо

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu