bambulab/BambuStudio #5224: 1.10.0.89 often not showing temperatures in device tab

dwmayr, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro 21H2

### Printer

Bambu A1 Mini

### How to reproduce

Just go to device tab - in some cases no temperatures are available, in most cases only hotend temperature are showed. If you'll leave this tab opened for a few minutes, it'll sometimes show the heatbed temperature.
Also it's showing chamber temperature, which A1 mini doesn't have.

### Actual results

[image]

### Expected results

Show proper hotend and heatbed temperature immediately.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17727972/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

dwmayr, 2024-11-13:
Also finish time is not working ;)

vieuxkiom, 2024-11-13:
Same issue with my X1C... with version 1.10.0.89 of bambu studio

Duudeeee, 2024-11-13:
same on new update version bambu

FallasAutomation, 2024-11-13:
Bug reported here:

https://github.com/bambulab/BambuStudio/issues/5196

Graystone55, 2024-11-14:
Verifying same has happened to me.  Studio sluggish at times and might be why it doesn't get temps.  I have P1P and X1C and both are missing temps at times.

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

walterwongbbl, 2024-11-21:
Hi.
please use 1.10.1 Public Release version.