bambulab/BambuStudio #5225: Cannot load Step Files - Crash

maCStacker, 2024-11-13:
### Bambu Studio Version

01.10.00.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S

### How to reproduce

try to insert a step file

### Actual results

Unhandled unknown exception; terminating the application

### Expected results

...

### Project file & Debug log uploads

not necessary

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-11-13:
please see: https://github.com/bambulab/BambuStudio/issues/5186

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu